brendan-w/python-OBD #10: Unable to Connect via Bluetooth - No Valid Ports Found

merktassel, 2015-01-15:
I'm having trouble accessing my ELM327 Bluetooth Adapter from python-OBD on Yosemite. I'm able to connect to it fine from third party programs like OBD Auto Doctor, but when using Scantool to find the correct port it returns an empty array. I've also tried manually passing in some ports I found by running ls /dev/tty.* with no luck. I've been trying various suggestions I found through Google, as well typical troubleshooting steps like restarting and reinstalling for the past couple of days, but I'm out of ideas as to what I could be doing wrong.

brendanwhitfield, 2015-01-15:
Thanks for bringing this up:

What little bluetooth information I have is actually from the upstream I forked from (which was linux). I don't have a mac, but here are my thoughts so far: 

+ The library this uses for serial is called pySerial, so I've been searching on that.

+ Other people that forked pyobd (the original library) say that it shows up on mac as `/dev/tty.usbmodem*`, or `/dev/cu.*`. If you're using a USB bluetooth dongle then it will probably be the former, and if your pairing it with the mac itself, it will probably be the later.

+ in rare occurences under linux, my user didn't have permissions to let pySerial use the ports. A quick and dirty way to test this is to run python as root with `sudo` (a terrible solution, but could be informative)

I'm going to try and get my hands on a friends mac later today

brendanwhitfield, 2015-01-15:
Also, if you haven't already, enable debug information with `obd.debug.console = True`, which will print out serial errors

merktassel, 2015-01-19:
Thanks for the quick response! I now know for sure that /dev/tty.OBDII-Port is the port that I should be using. However, running python with sudo and enabling debug information didn't make any difference. I also tried running the following script:

```python
import obd

obd.debug.console = True

connection = obd.OBD("/dev/tty.OBDII-Port")

cmd = obd.commands.RPM

response = connection.query(cmd)

print(response.value)
print(response.unit)
```

which returned

```
Starting python-OBD
Explicit port defined
Opening serial port...
Serial port successfully opened on /dev/tty.OBDII-Port
atz response: D?
ate0 response: ate0OK
Connected to ECU
querying for supported PIDs (commands)...
Sending command: 0100: Supported PIDs [01-20]
command returned: 41 00 80 40 00 01 41 00 BE 7F B8 13 41 00 80 00 00 01 
Receieved more data than expected, trying to parse anyways...
Sending command: 0120: Supported PIDs [21-40]
command returned: 41 20 80 01 80 01 41 20 80 07 E0 11 41 20 80 01 80 01 
Receieved more data than expected, trying to parse anyways...
Sending command: 0140: Supported PIDs [41-60]
command returned: 41 40 FE D0 00 00 41 40 40 00 00 00 41 40 40 00 00 00 
Receieved more data than expected, trying to parse anyways...
finished querying with 15 commands supported
========================================
'010C: Engine RPM' is not supported
None
None
```

brendanwhitfield, 2015-01-19:
That return data is quite peculiar...

This is not a serial/bluetooth problem. This is either a bug in the library, or a malfunctioning adapter. When you said it "connected" using OBD Auto Doctor, were you able to get accurate RPM data from the vehicle?

I should have time later tonight, and I'll write a script to help debug this.

merktassel, 2015-01-20:
Yes, I was able to get accurate data from the vehicle using OBD Auto Doctor. Today I tried hooking my adapter up to a different vehicle and running the script, and it worked. I then plugged the adapter back into my vehicle and ran the script, and it failed as it did before. I have a 2012 Chevrolet Sonic. What could be going on?

brendanwhitfield, 2015-01-20:
Found it!
Turns out, your chevy is modern enough that it has more than one ECU, so there may be multiple responses to one command. For instance (from your debug output):

```
41 00 80 40 00 01 41 00 BE 7F B8 13 41 00 80 00 00 01

# is actually a multiline response (the newlines got scrubbed somehow...)

41 00 80 40 00 01
41 00 BE 7F B8 13
41 00 80 00 00 01
```

The [ELM327 spec](http://elmelectronics.com/DSheets/ELM327DS.pdf) details multiline responses on page 42. It requires turning on some headers, and filtering the data. Implementing it in the library (correctly) will take fair bit of time and research, but if you need something now, I might be able to make a one-off patch to get you data from only the engine.

merktassel, 2015-01-20:
For now I'll see if I can get my hands on a different vehicle to test. But if it's something fixable from your end, I'd appreciate a patch as soon as you have time. Thanks for helping me out.

brendanwhitfield, 2015-01-21:
I made a branch called [multi-ecu](https://github.com/brendanwhitfield/python-OBD/tree/multi-ecu). It works on my single ECU car (and in the tests for multi-ECU cars), but if you get some time, I'd love to have you test it on your Chevy. Either clone or [download](https://github.com/brendanwhitfield/python-OBD/archive/multi-ecu.zip) the `multi-ecu` branch, and install it with `sudo python setup.py install` (preferably in a virtualenv). Run the same script as before, and if it doesn't work, the debug output should tell me what I need to know.

brendanwhitfield, 2015-02-08:
While I haven't been able to test the new system on a car with multiple ECUs, it is stable on single ECU cars. Since no existing functionality was lost, I have chosen to release to pypi. If you still wish to try it out, feel free to `pip install` the latest version.

merktassel, 2015-02-08:
Sorry I haven't had a chance to try it out yet. I've been busy and the submission deadline is coming up soon. After that, I should have plenty of time to test it and let you know what I find. Thanks for being so responsive.

merktassel, 2015-02-08:
I just updated to the latest release (0.3.0) and tried running the same script . I received the following output this time:

```
[obd] ========================== Starting python-OBD ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/tty.OBDII-Port'
[obd] Serial port successfully opened on /dev/tty.OBDII-Port
[obd] write: 'atz\r\n'
[obd] read() found nothing
[obd] read() found nothing
[obd] read: ''
[obd] Connection Error:
[obd]     atz (reset) did not return with an ELM version
[obd] Failed to connect
[obd] =========================================================================
[obd] '010C: Engine RPM' is not supported
None
None
```

brendanwhitfield, 2015-02-08:
Thank you! That tells me exactly what I need to know. For now, I'd recommend rolling back to version 0.2.0 while I hash this out.

```Shell
$ pip uninstall obd
$ pip install obd==0.2.0
```

I'm sorry it didn't work out of the box

merktassel, 2015-03-16:
Have you made any progress with this? I'd be happy to test it out and provide diagnostic information if needed.

brendanwhitfield, 2015-03-18:
Quite a bit of progress. A *lot* of code has changed, so when it comes time for release, I would love to have you test. Not quite done with it yet, so I don't want to waste your time, but it's nearing the finish line. I'll ping you I have something complete.

btw, Your previous debug data was a huge help in designing the new system. Thank you.

brendanwhitfield, 2015-04-11:
Ok, looks like master (81fe93b052) is stable. Feel free to try it out at your leisure.

bitmarker, 2015-04-15:
I have some troubles with connecting via bluetooth on my Mac. I could solve the problem by waiting 1 or 2 seconds after opening the serial port. Without this delay the communication seems not to work properly.

brendanwhitfield, 2015-04-19:
@bitmarker I have a hunch that this is a different issue than the one in this thread. Please open a new issue and post your debug output from the failure. Enable debug mode with `obd.debug.console = True`

brendanwhitfield, 2015-05-27:
@merktassel version 0.4.0 has all the completed work in it, and should (finally) solve this problem. Ok to close?

merktassel, 2015-05-27:
Have you had a chance to field test it yet? If not, I'd like to try it out tomorrow to confirm things are good to go. Otherwise, go ahead and close the issue. Either way, thanks for addressing it for me.

brendanwhitfield, 2015-05-27:
Yes, it's been tested on quite a few vehicles. Just letting you know, more than anything. Thanks for all your help on this!

closed 2015-05-27 by brendanwhitfield