brendan-w/python-OBD #11: feature request: non-OBD backends

xfxf, 2015-02-16:
Hi mate, nice library - have used it to prototype up a basic car dashboard.

I'm working on a project that will require me talking to different non-OBD but OBD like backends; ECU's like MegaSquirt, Mitsubishi MUT, etc.  I'd like a standardised interface to talk to all of them so I ask the question; do you consider it within the scope of this project to support non-OBD2 protocols or not?

Not asking for you to implement this, moreso will you accept patches? :)

brendanwhitfield, 2015-02-18:
Hi! Sorry for the delay, I've been thrashing around on this ever since you posted.

First, I'll admit that I have zero experience with other ECUs, so if any of my understanding is wrong, please point it out.

I would *love* to have a common API for all ECU types. However, I don't think python-OBD is the right place to do it (as tempting as it is). My concern is that other protocols/ECUs will eventually want to implement functions that go far beyond what we'd be able to write in a uniform API. For example, people might eventually want the engine tuning/tweaking capabilities of a different ECU, and that is best done with a dedicated library. Rather than implementing a small portion of it's functionality in OBD, I think that it might be worthy of having it's own module.

Since this is still an *awesome* idea though, here's a thought:

+ Implement seperate libraries. Feel free to fork me and overhaul it :-)
+ Create module to serve as the abstraction between the different ECUs. It would use our libraries as dependencies, and contain only the functionality *shared* by the various ECU types. If the programmer wishes to do more ECU specific operations, then he/she can use the underlying module on its own.

Not as clean as I'd like, since we'd be stacking APIs, but potentially avoids a mess. Since I think you know more than I do about this, I'd appreciate any input you have.

xfxf, 2015-02-23:
Interesting ideas.  I think i'll fork it and have a go at adding some abstraction.

I don't care to implement functions beyond monitoring (such as tuning); this really is the job of a specific library and not the job of something generic.  I really care about making a library which has a common interface for doing data logging and gauge style stuff for a whole bunch of different ECU's.

closed 2015-06-26 by brendanwhitfield