brendan-w/python-OBD #13: obdsim no longer works (ELM responded with unknown protocol)

charleswoolf, 2015-02-19:
A change between version 0.3.0 and the current commit (5ba24a32f1f74e2fc28c2ba0e1c3814a2dcfb6f1) broke the ability to use obdsim as an OBD simulator.

Here's a log with version 0.3.0:
```
Python 2.7.6 (default, Mar 22 2014, 22:59:56)
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
>>> obd.debug.console = True
>>> connection = obd.OBD("/dev/pts/7")
[obd] ========================== Starting python-OBD ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/pts/7'
[obd] Serial port successfully opened on /dev/pts/7
[obd] write: 'atz\r\n'
[obd] read: 'atz\r\n\r\nELM327 v1.3a OBDGPSLogger\r\n'
[obd] write: 'ate0\r\n'
[obd] read: 'ate0\r\n\r\nOK\r\n'
[obd] write: 'ath1\r\n'
[obd] read: '\r\nOK\r\n'
[obd] Connection successful
[obd] querying for supported PIDs (commands)...
[obd] Sending command: 0100: Supported PIDs [01-20]
[obd] write: '0100\r\n'
[obd] read: '\r\n7E8 06 41 00 FF FF FF FF FC\r\n'
[obd] Receieved less data than expected, trying to parse anyways...
[obd] finished querying with 24 commands supported
[obd] =========================================================================
```
Output from obdsim in this case:
```
SimPort name: /dev/pts/7                                   
Successfully initialised obdsim, entering main loop        
10.000261 seconds. 0 samples, 0 queries. 0.00 s/s, 0.00 q/s
Reset                                                      
Echo disabled                                              
Headers enabled                                            
10.001805 seconds. 1 samples, 4 queries. 0.10 s/s, 0.40 q/s
```

and here's the output with the latest commit (5ba24a32f1):
```
Python 2.7.6 (default, Mar 22 2014, 22:59:56)
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
>>> obd.debug.console = True
>>> connection = obd.OBD("/dev/pts/7")
[obd] ========================== Starting python-OBD ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/pts/7'
[obd] Serial port successfully opened on /dev/pts/4
[obd] write: 'ATZ\r\n'
[obd] read: 'ATZ\r\n\r\nELM327 v1.3a OBDGPSLogger\r\n'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\r\n\r\nOK\r\n'
[obd] write: 'ATH1\r\n'
[obd] read: '\r\nOK\r\n'
[obd] write: 'ATSP0\r\n'
[obd] read: '\r\nOK\r\n'
[obd] write: '0100\r\n'
[obd] read: '\r\n41 00 FF FF FF FF FC\r\n'
[obd] write: 'ATDPN\r\n'
[obd] read: '\r\n0\r\n'
[obd] Connection Error:
[obd]     ELM responded with unknown protocol
[obd] Failed to connect
[obd] =========================================================================
```
obdsim:
```
SimPort name: /dev/pts/7                                         
Successfully initialised obdsim, entering main loop   
10.000231 seconds. 0 samples, 0 queries. 0.00 s/s, 0.00 q/s           
Reset                                                            
Echo disabled                                                    
Headers enabled                                                  
New Protocol: Automatic                                          
10.000597 seconds. 1 samples, 5 queries. 0.10 s/s, 0.50 q/s      
10.000915 seconds. 0 samples, 1 queries. 0.00 s/s, 0.10 q/s 
```

In both cases obdsim was run just as `obdsim`. The version I'm using is 0.16.

I read the man pages for obdsim, and found that it can be told to emulate a specific ELM protocol (using `-p 4` for 'ISO 14230-4 (KWP 5BAUD)' for example), and these protocol versions appear to match with those defined in [elm327.py](https://github.com/brendanwhitfield/python-OBD/blob/master/obd/elm327.py#L52). However, this seems to have no effect (I think this may be being changed while obdsim is executing, as pointed out by 'New Protocol: Automatic' in its output).

Let me know if I can provide any more information to help with troubleshooting!

chrispyduck, 2015-02-19:
I'm working on a similar problem using Python 3 right now. See if the following helps:
port.__write, in port.write(), change "cmd" to "cmd.encode()"
port.__read, change result from [] to b''
port.__read, in the return statement, change "result" to "result.decode()"

That at least allows me to tx/rx messages, but it's not quite functional yet.

brendanwhitfield, 2015-02-20:
Thank you! I only recently started playing with obdsim, but ran into the same problem.

tl;dr
It works on an actual ELM327. Suspected obdsim problem. Found a workaround, but for anyone that's interested:

I'm hesitant to point at obdsim on this one, but it doesn't seem to be handling automatic protocol selection properly. Python-OBD leverages the ELM's "auto protocol" mode (trying to not reinvent the wheel here), and in doing so, uses the `AT SP 0` command. This *should* tell obdsim to "search" for a supported protocol upon sending the first command. The [ELM327 spec](http://www.elmelectronics.com/DSheets/ELM327DS.pdf) (page 36, lower right) mentions this, and states that subsequent `AT DPN` commands should return the ELM's protocol-of-choice. However, obdsim returns zero, which isn't a protocol...

My solution is to use a variation of the `AT SP 0` command that specifies a default protocol to try first. `AT SP A8` tells the ELM to try protocol 8 (ISO 15765-4, 11 bit, at 250k), and to perform a search only if that fails. This might actually be better than using `AT SP 0`, since all cars past 2008 are required to support CAN, so we might have better chances at avoiding an extensive search. 

I studied some of the obdsim source, but it'll take me a little longer to actually get to the bottom of this.

charleswoolf, 2015-02-23:
I can confirm that the current commit (d0cb3219ef8db98009d771c92028963835662067) has got obdsim working once again! See https://gist.github.com/charleswoolf/c17c90cdd180a6786545 for what I could get from it. As obdsim functionality has returned I'm happy for this issue to be closed.

Thanks for the work!

closed 2015-02-23 by charleswoolf