brendan-w/python-OBD #19: Unable to connect OBD to mac

koukumar, 2015-05-14:
I am trying to connect my mac to ODB which is connected to "Maruti Swift VDI".

```
import obd
obd.debug.console = True
import obd
connection = obd.OBD("/dev/tty.OBDII-SPP")
cmd = obd.commands.RPM
response = connection.query(cmd)
print(response.value)
print(response.unit)
```

```
[obd] ========================== python-OBD (v0.4.0) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/tty.OBDII-SPP'
[obd] Serial port successfully opened on /dev/tty.OBDII-SPP
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] __read() found nothing
[obd] __read() found nothing
[obd] __read() never recieved prompt character
[obd] read: ''
[obd] write: 'ATE0\r\n'
[obd] __read() found nothing
[obd] __read() found nothing
[obd] __read() never recieved prompt character
[obd] read: ''
[obd] Connection Error:
[obd]     ATE0 did not return 'OK'
[obd] Failed to connect
[obd] =========================================================================
[obd] '010C: Engine RPM' is not supported
None
None
```

But i was able to successfully connect it via Torque mobile app. The protocol is used was "iso14230-4 KWP fast". Can you please let me know if this is supported?

brendanwhitfield, 2015-05-16:
Try instantiating the connection with the baudrate 10400

```python
connection = obd.OBD("/dev/tty.OBDII-SPP", baudrate=10400)
```

koukumar, 2015-05-19:
Thanks for the suggestion. Even that didn't work. Wondering would there be any problem with mac SPP?

brendanwhitfield, 2015-05-20:
Sounds like it, python-OBD isn't getting *any* data from the device, so it's probably a setup issue. I've never used the mac bluetooth settings, but I believe you just need a simple RS-232 device.

koukumar, 2015-05-21:
Sure. Let me try using a different device and will update. Thanks for the reply.

brendanwhitfield, 2015-06-16:
any luck?

koukumar, 2015-06-16:
From mac it is still not working. I moved on to Linux based machines.

brendanwhitfield, 2015-06-21:
ok to close?

koukumar, 2015-06-22:
Yeah sure

koukumar, 2015-06-22:
Thanks for the help and follow up :)

closed 2015-06-26 by brendanwhitfield