brendan-w/python-OBD #20: is_connected() is false even though we're connected to the ELM

HawtDogFlvrWtr, 2015-05-21:
in previous versions, is_connected was true if we were connected to the elm device. Now it appears that is_connected only returns true if 0100 returns successfully. This allowed me to determine if we were successfully connected to the bluetooth/serial device, at which point a quick check for RPM determined if the vehicle was actually running or not.

brendanwhitfield, 2015-05-21:
Yes, 0.4.0 is the first version to actually check that the car returns data, but I'm not sure I understand your problem. I've never encountered a case where an RPM ping returns, but a 0100 does not. Here's how it should behave:

```python
# car OFF, engine OFF
connected = False   # Yes, in versions <= 0.3.0, this returned True.
rpm = None

# car ON, engine OFF
connected = True
rpm = 0

# car ON, engine ON
connected = True
rpm > 0
```

I found the old car OFF & engine OFF `is_connected()` response to be misleading. I figured, what's the point of a connection that can't recieve basic data like 0100. Am I wrong?

HawtDogFlvrWtr, 2015-05-21:
I was using it to determine if the OBD connection had been made, and then did a basic check to determine if the engine was started.

My suggestion would be to extend it and maybe add a is_engine that waits for a response from 0100 to determine if the engine is started, rather than rolling both into one. This allows for greater flexibility, because (as in my case) some may be using your library to detect engine status and potentially push other commands to the vehicle, such as remote start/stop/lock/unlock. With version 2, I was able to determine if the OBD was connected with is_connected, then loop until rpm returned a value to determine if the engine was running. During said loop of the engine being down, I'm able to push commands to my remote start to start the vehicle.

So basically, by making is_connected both engine and obd connection status, it's eliminated the ability for  me to perform remote functions on the vehicle. I can probably find a way around this, but it just makes more sense to break those two checks into separate functions.

Just my 2 cents though...

https://github.com/HawtDogFlvrWtr/uhacknect

brendanwhitfield, 2015-05-27:
Sorry for the delay,

I'm thinking about creating a status flag, which could have values such as:

- `UNCONNECTED` for no ELM connection
- `ELM_CONNECTED` for an ELM connection with no 0100 response
- `CAR_CONNECTED` for a working connection with the car

There's a bunch of details I still have to work out, such as handling the auto scanning of commands (when the initial connection did *not* return 0100). It might be a little while before I get this fully implemented.

Until this happens, I'd recommend trying out the [`obd.scanSerial()`](https://github.com/brendanwhitfield/python-OBD/wiki/OBD-Connections) utility. This might solve your problem of probing for the adapter. I'd like the behavior of `is_connected()` to stay the way it is for now, since that's the way most people use it.

btw: where did you find those remote start/stop codes? I've never heard about doing that through OBD before.

HawtDogFlvrWtr, 2015-05-27:
I will look into that. As for how I figured it out, one day I was sniffing and noticed that when I hit my fob it was registering, so I replayed and it worked.

HawtDogFlvrWtr, 2015-05-27:
ok, so scanSerial() works for me as a work around for now. I get the len of the list and if greater than 0, i assume i have a connection to the port that was found.

closed 2015-06-08 by HawtDogFlvrWtr