brendan-w/python-OBD #22: ELM responded with unknown protocol with version 0.4.0

BertrandBrelier, 2015-06-13:
Hello everybody,

I am trying to use python-obd to get data from the ELM327 but I have the following issue:

import obd
ports = obd.scanSerial()
print ports
['/dev/rfcomm0']
connection = obd.OBD(ports[0])
[obd] Connection Error:
[obd]     ELM responded with unknown protocol

Could you please help me fixing this issue ?
Let me know if you need information and/or performing tests.

Thank you,

Bertrand

brendanwhitfield, 2015-06-14:
Sure,
Please re-run your code, but with console debug enabled:

```python
...
obd.debug.console = True
connection = obd.OBD(ports[0])
...
```

That will show us all of the tx/rx data with the port.

BertrandBrelier, 2015-06-14:
Hello Brendan,

Thanks for your help.

Here is the output:

>>> import obd
>>> ports = obd.scanSerial()
>>> obd.debug.console = True
>>> print ports
['/dev/rfcomm0']
>>> connection = obd.OBD(ports[0])
[obd] ========================== python-OBD (v0.4.0) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/rfcomm0'
[obd] Serial port successfully opened on /dev/rfcomm0
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: 'ATZ\r\r\rELM327 v2.1\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\rOK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATSPA8\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'SEARCHING...\rUNABLE TO CONNECT\r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: '0\r\r'
[obd] Connection Error:
[obd]     ELM responded with unknown protocol
[obd] Failed to connect
[obd] =========================================================================

I am using linux (ubuntu 14.04), with a power2 bluetooth diagnostic scanner.

Thanks for your help,

Cheers,

Bertrand

brendanwhitfield, 2015-06-15:
python-OBD appears to be communicating with the ELM just fine, however, the ELM claims that it can't make a connection with your car. This could be anything from a simple electrical problem, to a faulty adapter (unlikely). Has this adapter worked for you in the past?

BertrandBrelier, 2015-06-15:
Hello Brendan,

This is the first time I use this adapter. Any suggestions to test it ?

Thanks,

Bertrand

BertrandBrelier, 2015-06-15:
Hello, 

I have installed ELM327 OBD terminal app on my smarphone.
Which command should I use to test the connection with my car ?

Thanks for your help, I really appreciate.

Cheers,

Bertrand

brendanwhitfield, 2015-06-15:
Try sending a `0100` command. If it works, this will return a bunch of hex. If not, it will say `UNABLE TO CONNECT`

BertrandBrelier, 2015-06-15:
Thanks Brendan.

I tried the command 0100 and it returned UNABLE TO CONNECT.
I tried on my car and my neighbour’s car and it failed with both cars.

Any suggestions ?

Thanks again for your help, I really appreciate.

Cheers,

Bertrand

brendanwhitfield, 2015-06-16:
Well, there's not a whole lot I can do at this point. If I had to guess, I'd say it's an electrical problem with the OBD-II connector. The ELM chip is responding normally. If it's not worth the money to return it, you might as well open it up and have a look inside. If you're lucky, it's just a couple bad solder joints.

BertrandBrelier, 2015-06-16:
I tried a few other commands in case it helps understanding the issue with my device:

ATZ
ATZ ELM327 v2.1
ATDP
ATDP ISO 15765-4 (CAN 11/500)

Thanks for your help,

brendanwhitfield, 2015-06-21:
The only thing curious about that output is the `ATDP ISO 15765-4 (CAN 11/500)`, but I think that's just the default. If you really want, you can try:

```
ATZ
ATDP
0100
ATDP
```

BertrandBrelier, 2015-06-21:
Here is what I got:

ATZ ELM327 v2.1
ATDP ISO 15765-4 (CAN 11/500)
0100 SEARCHING ... UNABLE TO CONNECT
ATDP ISO 15765-4 (CAN 11/500)

Thanks for your help,

brendanwhitfield, 2015-06-21:
hmm, doesn't reveal anything new. Sorry, but don't think there's anything more I can do. I'd still recommend looking for electrical problems at/inside the adapter.

BertrandBrelier, 2015-06-21:
Thanks Brendan,

I really appreciate your help. I have ordered another adapter.

Cheers,

BertrandBrelier, 2015-06-28:
Hello Brendan,

I bought another adapter and this one works fine with my smartphone.
Here is what I got with my smartphone:
ATRV 15.0
ATDP AUTO,ISO 15765-4 (CAN 11/500)
0100 7E8064100BE1FA813 7E906410098188013
010A NO DATA
010C 7E804410C1036 00
010D 7E804410D00  800
0104 7E804410446 800

With my laptop, I was able to pair the device:
hcitool dev
Devices:
        hci1    5C:F3:70:60:DE:41
        hci0    C4:85:08:56:4F:18
sudo hciconfig  hci1 up
hcitool scan 
Scanning ...
        88:18:56:68:98:EB       OBDII
sudo echo 1234|bluez-simple-agent hci1 88:18:56:68:98:EB
RequestPinCode (/org/bluez/1067/hci1/dev_88_18_56_68_98_EB)
Enter PIN Code: Release
New device (/org/bluez/1067/hci1/dev_88_18_56_68_98_EB)

sudo rfcomm connect 1 88:18:56:68:98:EB 1

rfcomm
rfcomm0: C4:85:08:56:4F:18 -> 88:18:56:68:98:EB channel 1 closed [reuse-dlc release-on-hup ]
rfcomm1: 5C:F3:70:60:DE:41 -> 88:18:56:68:98:EB channel 1 connected [reuse-dlc release-on-hup tty-attached]

>>> import obd
>>> obd.debug.console = True
>>> ports = obd.scanSerial()
>>> print ports             
['/dev/rfcomm1']
>>> connection = obd.OBD(ports[0])
[obd] ========================== python-OBD (v0.4.0) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/rfcomm1'
[obd] Serial port successfully opened on /dev/rfcomm1
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: '\r\rELM327 v2.1\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\rOK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATSPA8\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'SEARCHING...\rUNABLE TO CONNECT\r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: '0\r\r'
[obd] Connection Error:
[obd]     ELM responded with unknown protocol
[obd] Failed to connect
[obd] =========================================================================
>>> 

Am I doing something wrong when I paired the device ? 

Thanks,

BertrandBrelier, 2015-06-28:
After rebooting my laptop, I had something different:
>>> import obd
>>> ports = obd.scanSerial()
>>> print ports
['/dev/rfcomm1']
>>> obd.debug.console = True
>>> connection = obd.OBD(ports[0])
[obd] ========================== python-OBD (v0.4.0) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/rfcomm1'
[obd] Serial port successfully opened on /dev/rfcomm1
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] __read() found nothing
[obd] __read() found nothing
[obd] __read() never recieved prompt character
[obd] read: ''
[obd] write: 'ATE0\r\n'
[obd] __read() found nothing
[obd] __read() found nothing
[obd] __read() never recieved prompt character
[obd] read: ''
[obd] Connection Error:
[obd]     ATE0 did not return 'OK'
[obd] Failed to connect
[obd] =========================================================================

Thanks for your help,

brendanwhitfield, 2015-06-29:
Looks like rebooting the laptop lost the connection somehow. Whenever it shows `__read() never recieved prompt character` it means it can't recieve any data, which probably isn't a python problem.

Very confusing output though. I just made a branch called `bertrand` with a possible fix. Please `fetch` it and try it out at your leisure, (and copy the debug data please).

Thanks

BertrandBrelier, 2015-07-01:
Hello Brendan, 
I really appreciate your help.

I installed the Bertrand branch and performed the same test:

sudo rfcomm connect 1 88:18:56:68:98:EB 1
Connected /dev/rfcomm1 to 88:18:56:68:98:EB on channel 1

rfcomm
rfcomm1: 5C:F3:70:60:DE:41 -> 88:18:56:68:98:EB channel 1 connected [reuse-dlc release-on-hup tty-attached]

>>> import obd
>>> obd.debug.console = True
>>> ports = obd.scanSerial()
>>> print ports   
['/dev/rfcomm1']
>>> connection = obd.OBD(ports[0])
[obd] ========================== python-OBD (v0.4.0) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/rfcomm1'
[obd] Serial port successfully opened on /dev/rfcomm1                                                                                                                                                                            
[obd] write: 'ATZ\r\n'                                                                                                                                                                                                           
[obd] wait: 1 seconds                                                                                                                                                                                                            
[obd] read: 'ATZ\r\r\rELM327 v2.1\r\r'                                                                                                                                                                                           
[obd] write: 'ATE0\r\n'                                                                                                                                                                                                          
[obd] read: 'ATE0\rOK\r\r'                                                                                                                                                                                                       
[obd] write: 'ATH1\r\n'                                                                                                                                                                                                          
[obd] read: 'OK\r\r'                                                                                                                                                                                                             
[obd] write: 'ATL0\r\n'                                                                                                                                                                                                          
[obd] read: 'OK\r\r'                                                                                                                                                                                                             
[obd] write: '0100\r\n'                                                                                                                                                                                                          
[obd] read: 'SEARCHING...\r7E8 06 41 00 BE 1F A8 13 \r7E9 06 41 00 98 18 80 13 \r\r'                                                                                                                                             
[obd] write: 'ATDP\r\n'
[obd] read: 'AUTO, ISO 15765-4 (CAN 11/500)\r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: 'A6\r\r'
[obd] Connection successful
[obd] querying for supported PIDs (commands)...
[obd] Sending command: 0100: Supported PIDs [01-20]
[obd] write: '0100\r\n'
[obd] read: '7E8 06 41 00 BE 1F A8 13 \r7E9 06 41 00 98 18 80 13 \r\r'
[obd] Sending command: 0120: Supported PIDs [21-40]
[obd] write: '0120\r\n'
[obd] read: '7E8 06 41 20 90 07 E0 11 \r7E9 06 41 20 80 00 00 01 \r\r'
[obd] Sending command: 0140: Supported PIDs [41-60]
[obd] write: '0140\r\n'
[obd] read: '7E8 06 41 40 FA DC 80 00 \r7E9 06 41 40 CA 00 00 00 \r\r'
[obd] finished querying with 37 commands supported
[obd] =========================================================================

Thank you very much for your help.

Cheers,

Bertrand

BertrandBrelier, 2015-07-01:
For my personal understanding, may I ask what you have changed in the branch ?

Thank you,

Cheers,

Bertrand

brendanwhitfield, 2015-07-02:
I can't say I fully understand what's going on with your adapter, but here's what I did to solve it ([and here's the diff](https://github.com/brendanwhitfield/python-OBD/commit/eb789f94d3e4d4cd4d132e7ff17656c7ec77b1d3))

My library puts the adapter in "auto-protocol" mode using `AT SP A8`. After sending the first command (I use `0100`), the adapter is supposed to search/auto-detect which protocol is available. Once that's done, subsequent `AT DPN` queries are *supposed* to return the chosen protocol. In your case, the correct response (chosen protocol) would have been `A6`.

`AT DPN` did not return `A6`. Instead, it returned `0`, which is defined as "auto protocol". Technically, a valid answer, but doesn't help when we need to know which one it chose. 

Since the problem involved auto-protocol selection, I figured the only thing I've been doing differently (compared to your phone tests) is setting an auto protocol with `AT SP A8`. The branch I made for you removes this startup command, and hardwires the protocol choice to `A6` as reported from your phone.

I remember reading somewhere, in someone elses code, that the ELM's auto protocol feature is handy, but doesn't always work perfectly. This is the first time I've run into it, so I guess I need to start trying protocols manually with `AT TP` to get some definitive answers. I'll see what I can do.

In the meantime, feel to use the test branch I made. Fair warning though, hardwiring the protocol selection means it might only work on your car.

zaros, 2015-07-30:
Hi - I had the same issue and now can connect to my ELM327 adapter using the bertrand branch.  I think it would help if we can specify the protocol to use when creating the connection.

brendanwhitfield, 2015-07-30:
The user shouldn't have to worry about which protocol it uses, the library should be smart enough to abstract that. There's a fix, I just haven't had time to implement it.

The details:

Instead of relying on the ELM's "auto protocol" mode (which doesn't always pan out), protocols can be tested one-by-one using the `TP` command. It's slower, but I've read it's more reliable, and I'm not too concerned with speed, since it only happens at startup.

brendan-w, 2016-03-23:
Auto protocol has been changed back to `ATSP0`, and detection is being done with `ATDPN` (with `ATTP` as a fallback). Slated for 0.5.0

closed 2016-03-23 by brendan-w

shlok97, 2018-11-23:
[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\xfc\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: b'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: b'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
Protocol Used:  6 ISO 15765-4 (CAN 11/500)
[obd.obd] Sending command: b'010C': Engine RPM
[obd.elm327] write: b'010C\r\n'
[obd.elm327] read: b'NO DATA\r\r>'

I've been facing a similar issue. My ELM327 sensor works fine with my Android app. But I'm getting this response from the Raspberry Pi. As per my understanding, the connection is not Giving after writing "0100" since it's unable to connect. 
I really request you to help me out with this.
Thanks
Shlok Kapoor

alistair23, 2018-11-23:
Your log shows that it does connect. Can you run this with the master branch instead on the installed package

IhsanAziz, 2021-03-21:
[image]
I am new to OBD. I'm having this issue. I am unable to run rpm and speed commands. kindly help me