brendan-w/python-OBD #25: KODI addon

ochorocho, 2015-07-26:
Hey ....

i'm trying to fold python-OBD into a KODI addon.
But i cant make it work in a KODI plugin.
it works in when i run it outside the KODI folder:

My code:
http://paste.ubuntu.com/11941826/

Log error:
http://paste.ubuntu.com/11941812/

Any idea howto get rid of this and make python OBD work in as a KODI addon?

I'm new to python, maybe i missed something.

Thanks!

brendanwhitfield, 2015-07-30:
It looks like you have a file called `obd.py` in the root of your project on the RPi (that wasn't committed to the repo). Since the file has the same name as the library, python loads the *file* `obd.py`, and explodes when non-existant functions/constructors start getting called. The simple solution is to either toss or rename `/home/pi/.kodi/addons/plugin.program.obd/obd.py`.

ochorocho, 2015-07-30:
Thanks. Seems like i missed the basics.
I will try it :+1:

ochorocho, 2015-08-04:
It worked! thanks again.
But still making baby steps ... :-)

closed 2015-08-04 by ochorocho