brendan-w/python-OBD #26: Byte Buffer Decode Error When Communicating at Wrong Baud Rate

willbuckner, 2015-08-02:
The correct baud rate for my car appears to be 115200. When I attempt to connect using the default baud rate of 38400, I get the following error:

```python
In [1]: import obd

In [2]: connection = obd.OBD('/dev/ttyUSB1')
[obd] ========================== python-OBD (v0.4.1) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/ttyUSB1'
[obd] Serial port successfully opened on /dev/ttyUSB1
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] __read() found nothing
[obd] __read() found nothing
[obd] __read() never recieved prompt character
[obd] read: '~f\xfa\x9e'
'~f\xfa\x9e'
---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-2-ac2f7c876d8a> in <module>()
----> 1 connection = obd.OBD('/dev/ttyUSB1')

/usr/local/lib/python2.7/dist-packages/obd/obd.pyc in __init__(self, portstr, baudrate)
     50
>:q01 00 07 E5 00
     51         debug("========================== python-OBD (v%s) ==========================" % __version__)
---> 52         self.__connect(portstr, baudrate) # initialize by connecting and loading sensors
     53         debug("=========================================================================")
     54

/usr/local/lib/python2.7/dist-packages/obd/obd.pyc in __connect(self, portstr, baudrate)
     74         else:
     75             debug("Explicit port defined")
---> 76             self.port = ELM327(portstr, baudrate)
     77
     78         # if a connection was made, query for commands

/usr/local/lib/python2.7/dist-packages/obd/elm327.pyc in __init__(self, portname, baudrate)
     99         # ---------------------------- ATZ (reset) ----------------------------
    100         try:
--> 101             self.__send("ATZ", delay=1) # wait 1 second for ELM to initialize
    102             # return data can be junk, so don't bother checking
    103         except serial.SerialException as e:

/usr/local/lib/python2.7/dist-packages/obd/elm327.pyc in __send(self, cmd, delay)
    299             time.sleep(delay)
    300
--> 301         return self.__read()
    302
    303

/usr/local/lib/python2.7/dist-packages/obd/elm327.pyc in __read(self)
    364
    365         # convert bytes into a standard string
--> 366         raw = buffer.decode()
    367
    368         # splits into lines

UnicodeDecodeError: 'ascii' codec can't decode byte 0xfa in position 2: ordinal not in range(128)

In [3]: exit
```

When I manually specify `115200`, I get:

```python
In [2]: import obd

In [3]: connection = obd.OBD('/dev/ttyUSB1', 115200)
[obd] ========================== python-OBD (v0.4.1) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/ttyUSB1'
[obd] Serial port successfully opened on /dev/ttyUSB1
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: 'ATZ\r\r\rELM327 v1.3a\r\r'
'ATZ\r\r\rELM327 v1.3a\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\rOK\r\r'
'ATE0\rOK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
'OK\r\r'
[obd] write: 'ATSPA8\r\n'
[obd] read: 'OK\r\r'
'OK\r\r'
[obd] write: '0100\r\n'
[obd] __read() found nothing
[obd] read: 'SEARCHING...\r7E8 06 41 00 FE 3F A8 13 00 \r7E9 06 41 00 98 1A 80 13 00 \r\r'
'SEARCHING...\r7E8 06 41 00 FE 3F A8 13 00 \r7E9 06 41 00 98 1A 80 13 00 \r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: 'A6\r\r'
'A6\r\r'
[obd] Connection successful
[obd] querying for supported PIDs (commands)...
[obd] Sending command: 0100: Supported PIDs [01-20]
[obd] write: '0100\r\n'
[obd] read: '7E9 06 41 00 98 1A 80 13 00 \r7E8 06 41 00 FE 3F A8 13 00 \r\r'
'7E9 06 41 00 98 1A 80 13 00 \r7E8 06 41 00 FE 3F A8 13 00 \r\r'
[obd] Sending command: 0120: Supported PIDs [21-40]
[obd] write: '0120\r\n'
[obd] read: '7E8 06 41 20 80 07 F0 11 00 \r7E9 06 41 20 80 01 80 01 00 \r\r'
'7E8 06 41 20 80 07 F0 11 00 \r7E9 06 41 20 80 01 80 01 00 \r\r'
[obd] Sending command: 0140: Supported PIDs [41-60]
[obd] write: '0140\r\n'
[obd] read: '7E9 06 41 40 C0 80 00 00 00 \r7E8 06 41 40 FE D0 8C 81 00 \r\r'
'7E9 06 41 40 C0 80 00 00 00 \r7E8 06 41 40 FE D0 8C 81 00 \r\r'
[obd] finished querying with 41 commands supported
[obd] =========================================================================

In [4]:
```

brendanwhitfield, 2015-08-03:
yeah, some adapters take different bauds, and the current version can't automagically determine them (yet), so you'll have to specify it manually for now. I'll make a ticket for including some baudrate testing logic in 0.5.0

Thanks for the output, sorry it was a mess.

brendan-w, 2016-04-14:
Relevant notes for auto baudrate:

[ELM327 Data Sheet](http://elmelectronics.com/DSheets/ELM327DS.pdf)(pages 13 and 50)
I also found an *awesome* implementation of this [here](https://github.com/lukevp/Python-OBD-Scanner/blob/master/pyobd2-0.4/obd/interface/elm.py#L188-L230) (also GPLd)

Mark-Hetherington, 2016-04-14:
Ok, that is *far* more comprehensive than what I had imagined :)

brendan-w, 2016-07-04:
implemented baud rate detection on `devel`

closed 2016-07-04 by brendan-w