brendan-w/python-OBD #27: Changing ELM Protocol Detect Command

taylor224, 2015-09-15:
Hello,

I used this library to develop ECU Data Logger.

I am using ELM327 v2.1 Bluetooth Module and my car is Honda CR-V 2008.

First I am try to connect ECU with this library, It is failed likes this issue(https://github.com/brendanwhitfield/python-OBD/issues/22).

To find reason to fail connect, I turned on debug console in python-OBD.

[image]

The error occurred at 'ATSPA8' command.
After this command, '0100' failed cause of 'ATSPA8' command.
The failed message at 'ATSPA8' means, Protocol handshake failed.
I searched about 'ATSPA8', but I can't find document about this command.
So I changed this command to 'ATSP0'.
'ATSP0' means automatic Protocol handshake. It works with my ECU.
If 'ATSP0' works in other ECU, It is better to change 'ATSPA8' to 'ATSP0' in [https://github.com/brendanwhitfield/python-OBD/blob/master/obd/elm327.py].

[image]

brendanwhitfield, 2015-09-23:
It actually used to be an `ATSP0` command, but some OBD software emulators [don't support that command correctly](https://github.com/brendanwhitfield/python-OBD/issues/13). In fact, I recently learned that even some cars/adapter combinations have [trouble with it](https://github.com/brendanwhitfield/python-OBD/issues/22). I've been peeking at some other OBD implementations, and some have found that manually testing protocols with `TP` is more reliable. That's hopefully going to be implemented in `0.5.0`.

brendan-w, 2016-03-23:
Auto protocol has been changed back to `ATSP0`, and detection is being done with `ATDPN` (with `ATTP` as a fallback). Works with software obdsim, and other vehicles. Slated for 0.5.0

closed 2016-03-23 by brendan-w