brendan-w/python-OBD #28: Fast Refresh Command

taylor224, 2015-09-15:
Hello,

I found increase refresh rate of OBD command.

According to this article, If I append '1' tail of the command, can reduce response delay of query.
(http://stackoverflow.com/questions/21334147/send-multiple-obd-commands-together-and-get-response-simultaneously)

I tested this command with python-OBD custom query with Async force send, I saw querying time is reduced.

``` python
def rpm(_hex):
    v = unhex(_hex) # helper function to convert hex to int
    v = v / 4.0
    return (v, obd.Unit.RPM)

def speed(_hex):
    v = unhex(_hex)
    return (v, obd.Unit.KPH)

commFastRPM = OBDCommand("RPM", "Engine RPM", "01", "0C1", 2, rpm)
commFastSpeed = OBDCommand("SPEED", "Speed", "01", "0D1", 1, speed)

connection = obd.Async('/dev/rfcomm0')
connection.watch(commFastSpeed, force=True)
connection.watch(commFastRPM, force=True)
connection.start()

while(True):
    speeddata = connection.query(commFastSpeed).value
    rpmdata = connection.query(commFastRPM).value
    print 'Speed : ' + str(speeddata) + '\nRPM : ' + str(rpmdata) + '\n'
```

But It is still slow to use fast refresh application.

Here is analysis data I tested.
Async watching two command(RPM, SPEED)
First number is Timer time ( 0 second ~ unlimited )
Second number is actually updated data in that second
(62 - 7 is means 7 data received in 1 second).
In 1 second, OBD data refresh rate is 5~13 response/s (Before change to fast refresh command : 1~3 response/s).
```
62 - 7 =======
63 - 8 ========
64 - 6 ======
65 - 11 ===========
66 - 4 ====
67 - 7 =======
68 - 7 =======
69 - 9 =========
70 - 5 =====
71 - 5 =====
72 - 5 =====
73 - 8 ========
74 - 9 =========
75 - 9 =========
76 - 6 ======
77 - 5 =====
78 - 5 =====
79 - 5 =====
80 - 7 =======
81 - 9 =========
```

I want to get more fast refreshed data. Is there any way to get fast refreshed data?

Code Example from here:
https://gist.github.com/taylor224/f69b895f7fe01fe3f354

brendanwhitfield, 2015-09-23:
Hi @taylor224, thanks for the info!

You're right! The only problem is that using a `1` for every command could lead to race conditions, where there are multiple ECUs responding to the same command. I consulted the [docs](http://www.elmelectronics.com/DSheets/ELM327DS.pdf) (page 32), and that "parameter" can be used to return after `N` response lines. Since version `0.5.0` has an ECU table system in it, we could use the ECU count to determine how many lines to expect. We'll just have to be careful with multi-line responses.

Good catch, and thanks for the tests & data!

brendan-w, 2016-03-23:
has been implemented on `devel` as of b5ab69a. Slated for 0.5.0 release

closed 2016-03-23 by brendan-w