brendan-w/python-OBD #31: Request multiple PIDs during Async connections

brendanwhitfield, 2015-11-04:
[page 45](http://elmelectronics.com/DSheets/ELM327DS.pdf). Up to 6 PIDs can be sampled with a single command. This would be much more efficient. The multiline protocol handlers will need to be fully working, and it can only be done with the CAN (ISO 15765-4) protocol.

sommersoft, 2016-08-29:
I'm tagging in on this. I have to give back somehow, since your module is the best version I could find for my application. I've forked and created a branch "multi-command".

Python isn't my strongest language, but I've been doing this long enough to reference-manual my way through it. 🎯 

I'll keep ya's posted.

brendan-w, 2016-08-29:
Awesome, thanks! I'll take all the help I can get. I see you've using varargs; that's exactly what I was going to do.

Some notes:

- this only works over CAN, so it should use a test similar to [this one](https://github.com/brendan-w/python-OBD/blob/master/obd/obd.py#L231)
- we should be careful to only send commands that return a definite number of bytes (where `OBDCommands.bytes != 0`)
- the protocol decoders will need to be tweaked. They currently chop off the leading mode/PID bytes (I didn't know about multi-PID requests at the time, sorry). We'll need the PID byte(s) to match up data with the corresponding command. This won't be simple, as it will break some other existing code. I can handle this one for you (assume that `Message.data` holds `41 04 3F 05 ...` rather than `3F 05 ...`)

sommersoft, 2016-08-30:
Updated the error catching; asserts [CAN only](https://github.com/sommersoft/python-OBD/blob/multi-command/obd/obd.py#L253)

Spent the rest of the night digging through the code and the ELM sheet to figure out a good approach. From what I see so far, might be able to avoid "breaking" all of the decoders if a multi-PID specific function is used. 

Basically just fork out at the start of query(): `if len(cmds) > 1: if protocol == CAN: query_multi(cmds)`, then handle decoding in the CAN protocol functions by using the frame controls and response-length return message. A toggle variable could be used for higher-level state switching, if needed.

Just spit-balling and note taking; it's late.

brendan-w, 2016-08-30:
I'd like to leave OBDCommands (and associated decoders) operating on normal `Message` objects, with no global toggle vars for special handling. Fork out at the start of `query()` into the `query_multi()` function, and send the request. This will likely return a single `Message` object. To keep the existing code simple, I think it would make the most sense for `query_multi()` to split the received message into multiple, smaller `Message`s that can be passed to the existing OBDCommand/decoder setup as normal:

For instance

```
query_multi() sends:
    01 04 05 0B 0C

query_multi() receives:
    Message(41 04 3F 05 44 0B 21 0C 17 B8)

and splits into:
    Message(41 04 3F)
    Message(41 05 44)
    Message(41 0B 21)
    Message(41 0C 17 B8)

each of which is a valid message that can be handled by calling the associated OBDCommand.
```

Again: currently, the code base chops off the two beginning mode/PID bytes, but this should be changed anyway. The protocol handlers shouldn't touch the application layer.

sommersoft, 2016-09-07:
We are now sending and reading multiple commands! See logging below. The use of the vararg forces the commands to be entered directly into the function call, as opposed to a list/tuple. I couldn't get the function to unpack the list/tuple-in-a-tuple during the building of the command string. I'll keep working on this so that the single and multi functions have the same syntax (will also help the async call the query-multi function). **Do you think dropping the vararg and just using a list to pass the argument would work?**

Next major step is parsing the response. This will need the elm327/protocols response to include the line # and "inner" nulls so that we can arrange the lines properly. Or at least return the message in order if it doesn't already (still learning the lower layer functions). The query-multi function builds a dict to hold the commands, the bytes of the return (for parsing), and eventually the response to pass back to the caller.

Didn't think this would take me this long...had to learn various git things and trace all of python-OBD's function ladder. But, without further delay....success log!

> response = connection.query_multi(False, obd.commands.SPEED, obd.commands.RPM)

>[obd.obd] !=================== python-OBD_multi (v0.6.1) ===================!
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=auto
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.3a\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\r7E8 06 41 00 BE 3F E8 13 00 \r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A6\r\r>'
[obd.protocols.protocol] map ECU 0 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=115200 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'7E8 06 41 00 BE 3F E8 13 00 \r\r>'
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '0120\r\n'
[obd.elm327] read: b'7E8 06 41 20 90 07 F0 11 00 \r\r>'
[obd.obd] Sending command: 0140: Supported PIDs [41-60]
[obd.elm327] write: '0140\r\n'
[obd.elm327] read: b'7E8 06 41 40 FE DC 80 40 00 \r\r>'
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'7E8 06 46 00 C0 00 00 01 00 \r\r>'
[obd.obd] finished querying with 97 commands supported
[obd.obd]============================================================
**[obd.obd] Adding multi-command: 010D: Vehicle Speed
[obd.obd] Adding multi-command: 010C: Engine RPM
[obd.obd] cmd_string built: 01 0D 0C
[obd.elm327] write: '01 0D 0C\r\n'
[obd.elm327] read: b'7E8 06 41 0D 00 0C 00 00 00 \r\r>'** <---_key on/engine off; hence nulls_
**[obd.obd] Message rcvd: [obd.protocols.protocol.Message object at 0x13d1e70]** <---_return w/repr()?_
**[obd.obd] cmd_msg{}: {'0C': 2, '0D': 1}** <--- _dict for holding bytes to parse and return vals_

Quickjump: [query_multi](https://github.com/sommersoft/python-OBD/blob/multi-command/obd/obd.py#L293)

sommersoft, 2016-09-09:
Dropped the *arg, and now just passing a list. Works just as well.

Working on the return message parsing. No changes should be needed to protocols.

brendan-w, 2016-09-09:
Firstly: awesome! thank you!

Before we go into implementation details, let's nail down how the API *should* work (in a perfect world).

My gut tells me to use varargs because its allows people to specify commands directly (just like the normal `query()` function). In python, you can always pass a list of parameters as varargs like this:

```python
cmds = [obd.commands.RPM, obd.commands.SPEED]
connection.query_multi(*cmds)
```

For output, I think it makes the most sense to return a tuple, which can be iterated or unpacked. This enables syntax like this:

```python
rpm = connection.query_multi(obd.commands.RPM)
rpm, speed = connection.query_multi(obd.commands.RPM, obd.commands.SPEED)

# OR

cmds = [obd.commands.RPM, obd.commands.SPEED]
responses = connection.query_multi(*cmds)

for r in responses:
    # do stuff
```

as long as the order is preserved, I think that should be sufficient. `OBDResponse`s already contain references to the commands that triggered them, so I think returning a dictionary is redundant. If people want to "dynamically dispatch" the responses (like the `Async` class does), you could either:

```python
responses = connection.query(*cmds)

# rely on preserved indicies
cmds[0] # command
responses[0] # response

# OR

# lookup the command from the response
responses[0].command # command
responses[0] # response
```

---

On a side note:

It's tempting to roll all of this functionality into the existing query function (using varargs). But it creates an awkward problem where single commands produce a single element tuples. Not only is this not backwards compatible, but it'd be annoying, since doing simple things like

```python
connection.query(obd.commands.RPM)
# becomes
connection.query(obd.commands.RPM)[0]
```

odds are, if you're watching multiple commands, then you're already using the `Async` class. So for now, I'm fine with having two query functions: `query()` and `query_multi()`. Obviously, `Async` will eventually use `query_multi()` under the hood.

brendan-w, 2016-09-09:
I'd like to work on this with you, but GitHub isn't letting me send pull requests downstream to you. If you'd like my changes you can pull from my copy of your `multi-command` branch.

sommersoft, 2016-09-10:
> My gut tells me to use varargs because its allows people to specify commands directly (just like the normal  query()  function). In python, you can always pass a list of parameters as varargs like this:

This is one of those moments where my Python [in]experience makes me feel more of a hindrance than a help. Reading the varags documentation as much as I did, it wasn't until your comment that I realized I wasn't unpacking the call list when I sent it. doh! Definitely explains the list-in-a-tuple problem I was having. The only other thing against using the varargs I found was having to move the `force=?? `kwarg ahead of the `*cmds `when using python2.7 and below, and declare it in when calling the function. (my current application forces me to use 2.7 based on some other modules...still looking how to move it to python3)

I agree with the problems of rolling it all into query()... That was the experience I was having when I was trying to use query(*cmds), and only building the cmd_string in query_multi and passing it back.

I'll revert back to the vararg and unpack appropriately.

The next hurdle I'm currently figuring out is parsing the return message. It has occurred to me that since the ELM/ECU may not send the response back in the same order as requested, searching by order of commands sent may result in misread data.

```
cmds = RPM, SPEED, COOLANT_TEMP
msg_sent = 01 0C 0D 05
response = 41 0D05 0C0010 0510 # spaced to show cmd grouping
for cmd in cmds:
```
Once you got to COOLANT_TEMP (05), using str.find() will return "05 0C" instead of "05 10". So, my plan is to reverse my thinking and run through the response, find which command we're currently on and read appropriate byte positions, then remove it from the response string. Rinse and repeat.

Then each of them will be passed back in the requested order in a tuple. For now, I plan on keeping the cmd_msg{} to hold the parsed responses, then order them by *cmds into the return tuple. Thoughts?

brendan-w, 2016-09-10:
I'm a strong believer in learning via working on FOSS projects. As long as you're willing to learn, I'm happy to help :)

Regarding implementation/response parsing:

python-OBD's protocol layer (`obd/protocols/protocol_can.py`) already handles the "flattening" of response lines, length trimming, hex-checking, etc... Every Message object has a `.data` field that hold a bytearray of those parsed hex values, all put together. So we shouldn't reimplement that in `obd.py`.

Correct, we can't simply `find()` the PIDs/data, or assume that the data is in the correct order, or that its there at all. Luckily, we can rely on the structure of the response. Here's how it should work:

```shell
Message.data = 41 04 3F 05 44 0B 21 0C 17 B8 # this is a bytearray, using hex for clarity
                  04 3F 05 44 0B 21 0C 17 B8 # chop off mode number (save it for later)
#                 [   ] [   ] [   ] [      ] # here's how we want to split it
#                 ^^ the first byte is always a PID

#                 lookup the command with that PID, see how much data it expects
#                 in this case, PID 04: cmd.bytes = 1

                  04 3F 05 44 0B 21 0C 17 B8
#                 [   ] so now we know that this is a PID/data combo, meant for PID 04

#                 copy off that segment of data, and create a new message out of it:
                  Message.data = 41 04 3F # remember to add that mode byte back

#                 chop off the data we just read, and repeat:
                  05 44 0B 21 0C 17 B8
#                 ^^ first byte is once again, out next PID

# By the time you work your way through all of the bytes, you will wind up with a bunch of
# message objects, which can be sent to their respective commands for final value parsing:
Message.data = 41 04 3F
Message.data = 41 05 44
Message.data = 41 0B 21
Message.data = 41 0C 17 B8
```

---

NOTE: this *will* require a small change in the protocol layer, but, as promised, I'll handle that for you. For your own knowledge, here's why. Currently, the protocol handlers chop off the mode/PID bytes (thinking it would make things simpler downstream):

```shell
# what should be
Message.data = 41 04 3F 05 44 0B 21 0C 17 B8
# is currently
Message.data =       3F 05 44 0B 21 0C 17 B8 # missing mode and first PID byte
```

This was done because, up until now, only one command was ever in flight at a time, meaning I didn't need those bytes to figure out which command had been sent. The reason this is a problem is because we *will* need to know the PID of the first data snippet. Without this first PID byte, we will never be able to tell:

- which command the first data snippet is for
- how much data to chop off before we get to the next command

This change is not simple, as it breaks some other stuff (which is mostly why I've put off multi-PID for so long), but it's a change that *should happen anyway*. So I'm happy to do that part for you.

brendan-w, 2016-09-11:
Protocol tweaks are done and merged on to master. You now have access to all of the data (the whole application layer):

```python
Message.data = bytearray(b"41 04 3F 05 44 0B 21 0C 17 B8") # omitted 0x prefix for clarity.

# which can be accessed like a normal python list:
Message.data[0] == 0x41 == 65 
```

sommersoft, 2016-09-13:
So, I think I've worked it out in Python 3.5+ (got tired of writing in GitHub, pulling on the RPi, and running; so I switched to using IDLE on the WinBox). `byterarray.hex()` is a life saver here...it was much uglier in early versions.
  Let me know if I've re-constructed the `Message.data` bytearray incorrectly; I wasted/spent a lot of time figuring out bytes <--> str/int with the format I started in.

```
# simulate obd.py
from binascii import unhexlify, hexlify

cmd_msg = {'1D': 1, '0B': 1, '0C': 2, '2E': 4}
cmd_resp = {}
cmds = list(cmd_msg.keys())

raw = '411D040C01020B012E01020304'
messages = bytearray(unhexlify(raw))

pid_return = messages[0].hex
pid_return = '0' + pid_return[1:]

# simulation complete; parse the message
k = 0
for i in range(len(messages)):
    if k > 0:     # if k is > 0, we found a PID earlier
        k -= 1    # countdown as we step over a byte
        continue
    findme = messages[i:i + 1].hex().upper()   # get current PID
    if findme in cmds:
        k = cmd_msg[findme] # how many bytes is the response
        cmd_resp[(pid_return + findme)] = messages[i + 1: (i + k + 1)].hex() # pull response slice

```
And the result is:

```
Message.data = 41 1D 04 0C 01 02 0B 01 2E 01 02 03 04
Parsed into dict = {'010C': '0102', '011D': '04', '010B': '01', '012E': '01020304'}

```

Obviously, I'll need to change the output at the end to construct `Messages` , and merge into query_multi's structure. But that is a tomorrow thing...

brendan-w, 2016-09-13:
The output looks correct, but I think it'd be better if we did this without touching strings at all. This code makes use of the fact that OBDCommands can be used as keys in a dictionary. Here's what I was thinking:

*warning: not test. just wrote it*

```python
messages = self.interface.send_and_parse(cmd_string)
# etc...

master = messages[0] # the message that contains our response
mode = master.data.pop(0) # the mode byte (ie, for mode 01 this would be 0x41)

cmds_by_pid = { cmd.pid:cmd for cmd in cmds }
responses = { cmd:OBDResponse() for cmd in cmds }

while len(master.data) > 0:
    pid = master.data[0]
    cmd = cmds_by_pid.get(pid, None)

    # if the PID we pulled out wasn't one of the commands we were given
    # then something is very wrong. Abort, and proceed with whatever
    # we've decoded so far
    if cmd is None:
        break

    l = cmd.bytes - 1 # this figure INCLUDES the PID byte

    # if the message doesn't have enough data left in it to fulfill a
    # PID, then abort, and proceed with whatever we've decoded so far
    if l > len(master.data):
        break

    # construct a new message
    message = Message(master.frames) # copy of the original lines
    message.data = master.data[:l]
    message.data.insert(0, mode) # prepend the original mode byte
    responses[cmd] = cmd(message)

    # remove what we just read
    master.data = master.data[l:]
```

sommersoft, 2016-09-14:
Your approach definitely removes the need for Python3.5+. Which is a good thing. Based on my previous experience elsewhere, I tend to default to parsing in string-like fashion. Undoubtedly stepped out of my comfort zone on this one.

Even in it's "untested" state, it looks like it will run just fine. I'll drop it in, and see what we get.

sommersoft, 2016-09-14:
Got it running until `responses[cmd] = cmd(message)`, at which point it errors out as follows:
```
Traceback (most recent call last):
  File "test_mult.py", line 10, in <module>
    response = connection.query_multi(cmd) # send the command, and parse the response
  File "/obd/obd.py", line 398, in query_multi
    responses[cmd] = cmd(message)
  File "/obd/OBDCommand.py", line 89, in __call__
    messages = list(filter(for_us, messages))
TypeError: 'Message' object is not iterable
```

I have verified that it is grabbing the `pid`, `cmd`, and `byte` (L). Can't get a representation of the `master` or `message` other than the `message.data` just before the error is only "0x41". Which would (maybe?) explain why it won't iterate.

Verified that query() is still working fine. Only thing I changed was to add `from .protocol import Message` into the `protocols/__init__.py` so that Message could be called from query_multi().

brendan-w, 2016-09-15:
Sent you a pull request

sommersoft, 2016-09-15:
Git can be fun... I was only using the prints to figure out why the messages wouldn't iterate when trying to add to the responses{}.

I have it cycling all the way through, but I need to narrow down which of my two changes have really made a difference. Once again, that's a tomorrow thing.

sommersoft, 2016-09-17:
It's working with multiple commands. Guess it was the original assignment of `master` and looping through all message objects that was causing the `Not Iterable` errors.

Do we need to add some "empty" message return catching? Better explained, do we need to test that all requested commands got a response? I'm imagining a situation where say, the 2nd requested command isn't returned for some reason, it will place the wrong (3rd) value into the tuple at the 2nd position. Not sure how real this situation would be...just trying to think of ways it could break.

It's been a pleasure working on this with you. Thank you for your patience and helping me learn some more Python. It is definitely its own brand of language structure. Haha.

sommersoft, 2016-09-17:
Tried a couple things to break it a little and see how it handles some things. Only ran into one problem. I tried to call the following cmds:
`cmd = [obd.commands.SPEED, obd.commands.RPM, obd.commands[1][10]]`
`#[1][10] = MAF = '0110'`

And `test_cmd `returned `fuel pressure not supported`. Meaning that something converted `[1][10]` into `010A`... 

Did I call it wrong? Should I use `obd.commands[1]['10']` so that it doesn't convert to hex?

brendan-w, 2016-09-18:
Yes! Thank you! Always nice to meet people interested in the software, and making/learning things along the way is a bonus.

---

I don't think we need to test for all commands getting a response. The way it's set up now, it pre-populates `responses` with empty `OBDResponse`s, (which people already know about with the normal `query()` function). If a command's value was missing, then it will simply return the empty `OBDResponse`, which maintains the order.

The command lookup you mentioned is actually working like it's supposed to. Keep in mind that the request string `0110` is in *hex*, whereas the numbers you were using (`[1][10]`) are in *decimal*. So it's actually:

```python
obd.commands[0x01][0x10] = MAF = '0110'
# OR, in decimal
obd.commands[1][16] = MAF = '0110'
```

---

There's just a few things left now (mostly notes to self, I can finish it up from here if you want):

- Commands don't always have a known number bytes. Cases where `OBDCommand.bytes == 0` means that it may return an arbitrary number of bytes in response. Since multi-PID requests depend on known data lengths, we need to test for `all([ cmd.bytes > 0 for cmd in cmds ])` before continuing.
- I'm not sure iterating through the list of message is the right solution. The [ELM327 datasheet (pg. 45)](https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf), claims that all results will be held in a single message. That being said, there's always a chance that multiple ECU's will respond, and thus, will create multiple messages. We may need to do ECU flag checking to find out which message is the one for us.
- When fast mode is enabled (self.fast = True), different command strings can be built to speed things up:
  - sending an empty string (`b""`) will make the adapter resend the previous command. This helps speed up IO times.
  - not sure if this works in multi-query mode, but in single query, you can append a frame count to the request. After the adapter sees N frames, it will return the data immediately, rather than waiting for a timeout.
- Write proper tests

jagauthier, 2017-01-15:
Hey Guys,

  Just read through this thread, because I was wondering if this was possible.   It's been a few months, but I don't see any reference to this in the docs, or examples.   Is this functionality included?  (I actually did a pip install, but do not mind pulling directly from the source tree)

brendan-w, 2017-01-18:
Hey!

The core logic is there (a `query_multi()` function on the OBD object), but it never got finalized or used in Async. If you'd like to play with it, it's parked in a branch called `multi-command`. [Looks like I wrote docs for it on that branch too](https://github.com/brendan-w/python-OBD/blob/multi-command/docs/Connections.md#query_multicommands-forcefalse). One of the big blockers for me was testing, since (if memory serves), obdsim doesn't handle multiple requests.

jagauthier, 2017-01-18:
That would explain why I couldn't get it to work.  One of these would probably help!
https://github.com/carloop/simulator

I ended up polling supported bits out of PIDSA-C, and then querying those.  The results were pretty good, still thousands of requests over 10 seconds.