brendan-w/python-OBD #32: basic usage code need to run twice to get it work

enzofxxjia, 2015-11-04:
Hello Brendan 

I just want to test the basic usage code for my car. 
code is here 
```python
import obd

connection = obd.OBD("/dev/rfcomm0")
cmd = obd.commands.RPM # select an OBD command (sensor)

response = connection.query(cmd) # send the command, and parse the response

print(response.value)
print(response.unit)
obd.debug.console = True
```
every time I run this code, I need to run it twice 
The first time run, I got 
```python
[obd] '010E: RPM is not supported '
```
after that I close the program and run again 
It work prefect,I could read rpm in the python shell 
Any idea on how to fix that ?
Thanks

brendanwhitfield, 2015-11-04:
Hi! Please move the `obd.debug.console = True` line to just after `import obd`. That way, it will print all of the debug data from the connection process (which is probably where the problem is).

enzofxxjia, 2015-11-04:
Hi Brendan thanks for reply
I common out that code ```obd.debug.console = True```
But still need to run twice to get it print on the python shell. 
Do I need to put a delay between the 
```python 
response = connection.query(cmd) # send the command, and parse the response
```
and 
```python 
print(response.value)
```

brendanwhitfield, 2015-11-04:
Nope, don't comment anything out.

Here, run this script, and send me all of the output please:

```python
import obd

# needs to be set BEFORE you make a connection
obd.debug.console = True

connection = obd.OBD("/dev/rfcomm0")
cmd = obd.commands.RPM # select an OBD command (sensor)

response = connection.query(cmd) # send the command, and parse the response

print(response.value)
print(response.unit)
```

enzofxxjia, 2015-11-06:
This photo is the output of the first run 
[image]
something error with the ATE0 command 
than, I close the program and run again 
It pass this error 
[image]
[image]

I am using the raspberry PI put run this code 
However,if I use the my laptop (win10), I didn't get any error in the first run.

brendanwhitfield, 2015-11-07:
Are you using code from `devel`? Version 0.4.1 doesn't send `ATSP0`. If you are, word of warning: there may be a lot of things broken on `devel`. I only recently got basic functionality working.

Anyhow, it looks like the initial reset command didn't get transmitted properly (and caused one of the following commands to fail). If it works on your laptop, then it's probably an eccentricity with the Pi's bluetooth connection. It's not a big deal. I've made a fix on `devel` that will be released with 0.5.0. If you're not using devel, or if you cant wait, here's the diff that should fix the problem c091439.

closed 2016-03-23 by brendan-w