brendan-w/python-OBD #36: No data from python-OBD - '010C: Engine RPM' is not supported

emorfam, 2016-01-26:
I'm currently having problems to get any readings from python-OBD on Ubuntu 14.04 using an ELM327 Bluetooth OBD dongle. With Torque this dongle works fine. I tried the following script:
```python
import obd
obd.debug.console = True
connection = obd.OBD("/dev/rfcomm0")
cmd = obd.commands.RPM
response = connection.query(cmd)
print(response.value)
print(response.unit)
```
```
[obd] ========================== python-OBD (v0.4.1) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/rfcomm0'
[obd] Serial port successfully opened on /dev/rfcomm0
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: '\r\rELM327 v2.1\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\rOK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATSP0\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'SEARCHING...\rUNABLE TO CONNECT\r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: 'A0\r\r'
[obd] ELM responded with unknown protocol. Trying them one-by-one
[obd] write: 'ATTP6\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'NO DATA\r\r'
[obd] Connection successful
[obd] querying for supported PIDs (commands)...
[obd] Sending command: 0100: Supported PIDs [01-20]
[obd] write: '01000\r\n'
[obd] read: 'NO DATA\r\r'
[obd] Sending command: 0200: DTC Supported PIDs [01-20]
[obd] write: '02000\r\n'
[obd] read: 'NO DATA\r\r'
[obd] finished querying with 0 commands supported
[obd] =========================================================================
[obd] '010C: Engine RPM' is not supported
None
None
```
I also tried it with a USB dongle - the problem is the same. Switching to the devel branch of python-OBD also did not solve the issue. I also tried to set the baudrate explicity:
```python
connection = obd.OBD("/dev/rfcomm0", baudrate=10400)
```
Still no luck. Any ideas?

closed 2016-01-26 by emorfam

E-genin, 2018-05-30:
Did you solve the problem , I'm facing the same !!

banujan59, 2021-04-09:
@emorfam How did you solve it? I'm having the same problem.

spazzola, 2021-04-26:
Same problem here :/