brendan-w/python-OBD #37: Cannot connect to OBD dongle on devel branch

emorfam, 2016-01-26:
I try to read the battery voltage (AT RV) from an ELM327 USB OBD dongle using python-OBD on Ubuntu 14.04. As far as I understand it, reading the battery voltage is currently only supported on the devel branch. However, when I try to connect to the dongle using 
```python
import obd
obd.debug.console = True
connection = obd.OBD("/dev/ttyUSB0")
```
I get the following result
```
[obd] ========================== python-OBD (v0.4.1) ==========================
[obd] Explicit port defined
[obd] Opening serial port '/dev/ttyUSB0'
[obd] Serial port successfully opened on /dev/ttyUSB0
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: '\r\rELM327 v1.5\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'ATE0\rOK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATSP0\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'SEARCHING...\r7E8 06 41 00 98 3B A0 13 \r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: 'A6\r\r'
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-15-12e6daef3864> in <module>()
      3 
      4 # --- USB ---
----> 5 connection = obd.OBD("/dev/ttyUSB0")
      6 
      7 

/home/eric/anaconda/lib/python2.7/site-packages/obd/obd.pyc in __init__(self, portstr, baudrate, protocol, fast)
     54 
     55         debug("========================== python-OBD (v%s) ==========================" % __version__)
---> 56         self.__connect(portstr, baudrate, protocol) # initialize by connecting and loading sensors
     57         self.__load_commands()            # try to load the car's supported commands
     58         debug("=========================================================================")

/home/eric/anaconda/lib/python2.7/site-packages/obd/obd.pyc in __connect(self, portstr, baudrate, protocol)
     81         else:
     82             debug("Explicit port defined")
---> 83             self.port = ELM327(portstr, baudrate, protocol)
     84 
     85         # if the connection failed, close it

/home/eric/anaconda/lib/python2.7/site-packages/obd/elm327.pyc in __init__(self, portname, baudrate, protocol)
    142 
    143         # try to communicate with the car, and load the correct protocol parser
--> 144         if self.load_protocol():
    145             self.__status = OBDStatus.CAR_CONNECTED
    146             debug("Connection successful")

/home/eric/anaconda/lib/python2.7/site-packages/obd/elm327.pyc in load_protocol(self)
    180         if p in self._SUPPORTED_PROTOCOLS:
    181             # jackpot, instantiate the corresponding protocol handler
--> 182             self.__protocol = self._SUPPORTED_PROTOCOLS[p](r0100)
    183             return True
    184         else:

/home/eric/anaconda/lib/python2.7/site-packages/obd/protocols/protocol_can.pyc in __init__(self, lines_0100)
    280     ELM_ID = "6"
    281     def __init__(self, lines_0100):
--> 282         CANProtocol.__init__(self, lines_0100, id_bits=11)
    283 
    284 

/home/eric/anaconda/lib/python2.7/site-packages/obd/protocols/protocol_can.pyc in __init__(self, lines_0100, id_bits)
     48         # Protocol __init__ uses the parsing system.
     49         self.id_bits = id_bits
---> 50         Protocol.__init__(self, lines_0100)
     51 
     52 

/home/eric/anaconda/lib/python2.7/site-packages/obd/protocols/protocol.pyc in __init__(self, lines_0100)
    137         # parse the 0100 data into messages
    138         # NOTE: at this point, their "ecu" property will be UNKNOWN
--> 139         messages = self(lines_0100)
    140 
    141         # read the messages and assemble the map

/home/eric/anaconda/lib/python2.7/site-packages/obd/protocols/protocol.pyc in __call__(self, lines)
    178             # subclass function to parse the lines into Frames
    179             # drop frames that couldn't be parsed
--> 180             if self.parse_frame(frame):
    181                 frames.append(frame)
    182 

/home/eric/anaconda/lib/python2.7/site-packages/obd/protocols/protocol_can.pyc in parse_frame(self, frame)
     86             # 00 00 07 E8 06 41 00 BE 7F B8 13
     87 
---> 88             frame.priority = raw_bytes[2] & 0x0F  # always 7
     89             frame.addr_mode = raw_bytes[3] & 0xF0  # 0xD0 = functional, 0xE0 = physical
     90 

TypeError: unsupported operand type(s) for &: 'str' and 'int'
```
Any ideas what going on here? I tried it with a Bluetooth dongle as well - the problem is there too.

emorfam, 2016-01-27:
Using Python 3.5 instead of 2.7 on the devel branch solved the problem.

closed 2016-01-27 by emorfam

brendan-w, 2016-01-27:
I didn't think anyone was watching devel, so I've been considering it an unstable branch. Yeah, I hadn't sorted out the python2 problems yet. Latest commit (063ec83) should fix python2.

Beware of devel though, there may be a few dragons left down there.

emorfam, 2016-01-28:
Thanks for the info. Any plans to merge the voltage feature from devel to master in the future?

brendan-w, 2016-01-28:
Yeah, I'm slowly building up to a new release, but I didn't have an exact date in mind.