brendan-w/python-OBD #38: Error - "Failed to choose primary ECU"

bendrick92, 2016-03-29:
After utilizing 'bluetoothctl' to pair and bind my bluetooth OBD adapter to '/dev/rfcomm0', I get the following error when attempting to initialize an OBD connection:

"Connection Error"
"Failed to choose primary ECU"

I've followed the steps outlined in the documentation with no luck.  I'm running Raspbian Jessie Lite, which seems to utilize a different pairing process than typical.

brendan-w, 2016-03-30:
The computer successfully connected to your adapter, but the adapter couldn't connect to the car. When this happens to me, it's usually a hardware problem, like a flaky OBD-II connector, or forgotting to turn on the car.

Hope this helps, sorry about the not-so-helpful debug output

bendrick92, 2016-03-30:
Thanks for the quick reply!  I will try a different OBD-II adapter tonight and report back.  Some additional research has got me thinking it might be Bluetooth permissions in the new release of Raspbian.

brendan-w, 2016-03-30:
Doesn't look like a bluetooth problem. That line that says `SEARCHING...\rUNABLE TO CONNECT\r\r` is actually your adapter talking back to your computer, so it looks like the bluetooth is working perfectly. That message is only given when the adapter's OBD side can't find the car.

brendan-w, 2016-07-29:
Still having issues?

bendrick92, 2016-07-29:
Thanks for checking back!  I forgot about this.  Haven't seen the issue pop up in a few months now, so I'll consider it resolved!

closed 2016-07-29 by bendrick92