brendan-w/python-OBD #40: Steering wheel angle?

smilanko, 2016-03-30:
Looking at the original project, there are some references to the steering wheel angle. Is this not supported in this forked version? I cannot seem to find the command that can be invoked.

brendan-w, 2016-03-31:
Not that I know of. I grepped the original codebase and only found a couple error descriptions in the DTC tables:

```
obd2_codes.py
1918:	"U0126" : "Lost Communication With Steering Angle Sensor Module" , 
2056:	"U0328" : "Software Incompatibility with Steering Angle Sensor Module" , 
2088:	"U0428" : "Invalid Data Received From Steering Angle Sensor Module" , 

build/lib/obd/codes.py
1927:    "U0126" : "Lost Communication With Steering Angle Sensor Module",
2065:    "U0328" : "Software Incompatibility with Steering Angle Sensor Module",
2097:    "U0428" : "Invalid Data Received From Steering Angle Sensor Module",
```

I do not know how to access this sensor. As far as I (and [wikipedia](https://en.wikipedia.org/wiki/OBD-II_PIDs) know, it is not in the "standard" range of PIDs. It is possible that it is a manufacturer-specific command (something which python-OBD doesn't handle out of the box).

Good question though. I'd love to find out.

smilanko, 2016-03-31:
hmm it seems that you are right. I might have misread something. What's odd here is that some cars have a SAS (Steering wheel Angle Sensor) while others do not, so I am guessing its still not part of the standard/requirement. I am also unsure how to query for it as well to be honest. :/

brendan-w, 2016-06-30:
I found a [reference to this](http://read.pudn.com/downloads138/sourcecode/others/590670/SAE%20J2178_2v002.pdf) (pg. 17). It claims that Wheel Angle is PID `300E`. I'm not really clear on how standardized this is, or if it's accurate at all (it from 1999 after all), but it might be worth a shot.

smilanko, 2016-06-30:
if we could make that work, at least on some vehicles, you have no idea how many scholarly students/researchers we would be helping. I will take a look over the next few weeks, and maybe try out this code on different cars. I'll let you know what I come up with.

brendan-w, 2016-06-30:
More snooping turned up this one for Toyota: `2147`

kuon, 2016-11-20:
How would you try those codes?

I tried

```python

def angle(messages):
    """ decoder for angle messages """
    print(messages[0].data)
    print(messages[1].data)
    return 0

cmd = OBDCommand("Angle",
                 "Steering Wheel Angle",
                 b"2147",
                 0,
                 angle,
                 ECU.ALL,
                 False)

connection.supported_commands.add(cmd)
connection.query(cmd)

```

On my toyota, with 2106, 2147, 3001, 300E. But I don't know if it's the right thing to do.

I am trying to get the steering wheel position.

raphaottoni, 2017-05-07:
How is it going? Did someone find the solution for it?  @kuon Did your code work?

@smilanko  Thats is true! This would help a lot of researchers out there, myself inclued =)

kuon, 2017-05-08:
I ended up writing a small bit of C:

https://github.com/kuon/backup-camera/blob/master/src/canbus.c

The steering filter is hardcoded at line 27 and 28.

raphaottoni, 2017-05-08:
@kuon Thanks for the tip but i didnt follow =/ 

What does this code mean? In terms of python-obd.

> device_filter.can_id = 0x25;
> device_filter.can_mask  = 0x7ff;

Does it mean that the command PID for steering wheel on your car is 0x25?  How did u find that?

What is this mask?  

Is it possible to reproduce this with the coded u posted on 20 nov 2016 up here?

thanks

kuon, 2017-05-09:
In term of python-obd, I am not sure. But it is the filter for the socket. What it does, is that it will accept all messages (`can_mask`) from device `0x25`. So yes, in my car, the PID for steering is `0x25`.

I found this id here http://tucrrc.utulsa.edu/ToyotaCAN.html
i used `candump` from https://github.com/linux-can/can-utils to check for messages. When I verified that `0x25` was the correct ID, it was straightforward.

The thing is, `0x25` is the ECU id. And I have no idea how to get that with python-odb, as I gave up quite quickly.

Also, note that my code is hooked directly to CANBUS, it is not using ODB II. I hooked directly to the CANBUS via a [PICAN](http://skpang.co.uk/catalog/pican2-canbus-board-for-raspberry-pi-2-p-1475.html) on a raspberry pi.

I know that some cars (mine does), provide direct access to CANBUS via the ODB II port, and some USB adapters allow you to actually access the CANBUS directly. But I don't know how.

ergueta, 2019-01-29:
```
dde = OBDCommand("DDE",          
               "Driver's Demand Engine - Percent Torque",    
               b"0161", 
               1,         
               driverDemandEngine,        
               ECU.ENGINE, 
               True)             

response` = o.query(dde,force=True)
print(response.value)
```

I tried to make it run but also got as response "None", I'm very interested in torque information from obd2. Why can't I get this parameter from the custom command above? Is there another way to get it?

Thanks,
Felipe

selffactory, 2020-09-17:
I guess, at python-OBD, command <b"xxyy">, xx is Mode, yy is PID.
So, I want to get Steering Angle, too.
I will try <b"0625">. (Mode 06 - CAN protocol, PID 25 from [ToyotaCAN](http://opengarages.org/index.php/Toyota_CAN_ID)

DavidTMuc, 2021-06-17:
How is it going? Did anyone achieved to determine the steering wheel angle?

Thanks,
David

selffactory, 2021-06-18:
Stop the project now. I have never obtained any responce from my ZN6.