brendan-w/python-OBD #41: Serial port successfully opened, but then hangs

Dust-Buster, 2016-04-03:
Thanks – I couldn’t wait for 0.5.0…lol. So, I did the uninstall and git clone / fetch / checkout. Python 2.7 (more libraries) on Win7 Pro:

import obd # works – says it’s v0.4.1
obd.debug.console = True # works
connection = obd.OBD('\.\COM9') # says Serial port successfully opened, but then \n without giving the [obd] or python ">>>" command line prompt and hangs (for at least two minutes before I Ctrl-Break)

Any suggestions?

brendan-w, 2016-04-03:
Can you send me all of the debug output that showed up on your console please?

Dust-Buster, 2016-04-06:
Certainly - please see the attached. Oh! I did mention previously that I can read the ELM327 from the Android app Torque Pro. I also should have said, "Any further suggestions?" as you have already provided guidance.
[image]
Previously, I also did the following:
ports = obd.scanSerial() # works
print ports # works; “[ ‘\\.\\COM1’,’\\.\\COM9’] with Win7 Pro saying: Control Panel > Administrative Tools > Computer Management > Device Manager > Ports (COM & LPT) # says COM8 and COM9

brendan-w, 2016-04-12:
Sorry for the delay,

I can't say for certain, since it's not even throwing any errors, but my guesses are:
- Wrong port. You can always try `COM8` manually, even though it doesn't show up in `scanSerial()`
- Permissions issue. Try running as admin?

Looks like the port is blocking on the very first write, and I'm surprised it isn't timing out (it should, after 3 seconds). I'm away from my windows machine right now, but if I get time, I'll try it out.

Dust-Buster, 2016-04-13:
No problem

Surely this isn’t because I’m running python v2.7.11 instead of v3.5.1 or my BT device driver.

**Wrong Port: **
Win7 Pro: Control Panel > Administrative Tools > Computer Management > Device Manager > Ports (COM & LPT) # says COM8 and COM9
connection = obd.OBD(\.\COM8) # doesn’t work

**Permissions Issue:
**Win7 Pro: Start > Command Prompt > ‘Right-click’ “Run as Administrator” ‘Left-click’

Not sure if this helps any:
import serial
ser = serial.Serial('\.\COM9', 9600, timeout=1) # works
print ser # Serial<id=0x21db2d0, open=True>>port=’\\.\\COM9’, baudrate=9600, bytesize=8. parity=’N’, stopbits=1, timeout=1, xonxoff=False, rtscts=False, dsrdtr=False>
ser.write("\x0100 \r") # didn’t work, then hangs
ser.write("\x01 00 \r") # didn’t work, then hangs
ser.write("\x0100\r") # didn’t work, then hangs.
ser.write("\x01\x00\r") # doesn’t work
## I’m not getting to hex = ser.readline()

Thanks for your time.

Dust-Buster, 2016-05-04:
Shall I try using Hayes AT Modem Commands?

At this point, I don't know if the problem is Win7 Pro, the device driver, the BT internal device or the python version. The ELM327 obviously works (with Torque Pro on Android).

Where can I find out about Renault or more specifically the Continental (Siemens) SID 3xx (300 to 310) headers, filters and check sums?

brendan-w, 2016-07-04:
Wow, this issue fell off my radar, sorry about that.

Yes, if it's hanging with using pyserial directly, it's most likely a windows/bluetooth problem. Do either of those COM ports disappear when you disconnect from the adapter?

Dust-Buster, 2016-07-14:
No problem – I had to retrieve my ELM327 anyway from a “python” guy with a unix machine that detected that baud rate was actually 115200. He tried “ATZ”, “ATRV” and “ATSP0” on his vehicle. He said that this ELM327 expects OBD commands to be made up of hexadecimal codes written in ASCII characters, rather than sending the actual hex byte values.

I think WinPro7 is the problem. It doesn’t report the actual baud rate and because I went through the handshake exercise, anytime I turn the BlueTooth Adapter On, Devices and Printers brings up the OBDII avatar. Nothing changes if I walk out of BlueTooth range, until I switch off the adapter.

In python the com ports did disappear previously when turning off the adapter.

brendan-w, 2016-07-18:
I noticed you were using hex rather than ascii, but that shouldn't matter. Any unknown commands should at least return a `?\r\n>` from the scanner. Try specifying that baudrate manually:

```python
connection = obd.OBD('.\COM8', baudrate=115200)
```

The default was 38400, so that might explain it. Though, if the port still hangs, then it's a windows/port setup thing.

masudap, 2018-10-28:
i solved a closely similar to this problem in the following thread and it might help.
from my experience, the function scan_serial() try to check all the ports and some ports does not repose and the function indefinitely wait for a response. so if we set the return value of the above function that might work, at least it worked for me, the description i provided in the following thread.

https://github.com/brendan-w/python-OBD/issues/58

Dust-Buster, 2018-11-19:
> Regarding "failure to auto set baudrate", #122
> <https://github.com/brendan-w/python-OBD/pull/122> should hopefully
> resolves some of these inconsistent behaviors. Please take a look and let
> me know if this resolves any of these problems... it's a pretty small yet
> significant change to python-OBD.

Thanks

I need my programming nephew to help me with this as I've completely
forgotten what I was doing, but he has taken on a new job. Hopefully, he'll
have some time during the holidays.

Cheers

closed 2018-12-01 by alistair23