brendan-w/python-OBD #42: VIN Support

Fr3ak1n0ut, 2016-04-13:
Are there plans to add VIN Support to python-OBD (Code 0902)?
[https://en.wikipedia.org/wiki/OBD-II_PIDs#Mode_09](url)

brendan-w, 2016-07-04:
I saw you, don't worry. Will be working on this soon. Hasn't been at the top of my list, since none of my vehicles support this PID (they're old). Thus, I can't really test it, outside of examples in the standards docs. Once I get something implemented, I'll encourage people to test it out.

Fr3ak1n0ut, 2016-07-04:
Alright, thanks ;) 
I already implemented this on my own, might submit a pull request soon :)

brendan-w, 2016-07-04:
Sweet, please do!

kellogsgit, 2017-06-25:
I will work on this feature. I would also like to extract brand from VIN in order to be able to derive a new class for each brand.

brendan-w, 2017-06-26:
https://github.com/brendan-w/python-OBD/blob/master/obd/commands.py#L292

Have at it. Never been tested.

farshadz, 2017-10-27:
Dear Brendan:
I tested your code for VIN, unfortunately it is not working for my case and recieveing Get Vehicle information is not supported. After I search,I guess because my car using CAN bus (BMW Serie 1) and it return the data in multiple messages. I have not go through your code yet to find a solution. Just wanted to let you know about the test.

pmundt, 2019-05-28:
As there doesn't appear to have been any progress on this front, I assume the current status is still unchanged, and that the Mode 9 commands that were added and later removed from commands.py are where things were left off.

We have an OBD-II simulator that can utilize most of the available protocols, and are happy to test this out. As we would like to support VIN-retrieval via Mode 9 in our app, we'll have a go at extending what's already there and extending it for a couple of different protocols.

alistair23, 2019-05-28:
It doesn't look like anyone has worked on this.

That would be great if you could add support. I will be happy to review your pull requests and help get your work merged.

closed 2020-03-06 by alistair23