brendan-w/python-OBD #43: STN11xx support

Mark-Hetherington, 2016-04-14:
I'm just looking into if it is possible, or worthwhile, to support the STN11xx chip. From what I can see these are all ELM327 compatible, but using the STN protocol might allow for better performance or other features

brendan-w, 2016-07-27:
Note to self: [datasheet](http://www.kds-online.com/Downloads/OBD/OBDLink_STN11xx-ds.pdf)

Yup, most of it looks cribbed from the 327. When I get some time, I'll give it a read.

kellogsgit, 2017-06-27:
Please tell me how to buy this adapter and I will work on supporting it. Could be very interesting.

kellogsgit, 2017-06-27:
The new chip recommended for new designs is STN 2120
http://www.obdsol.com/solutions/chips/

kellogsgit, 2017-06-27:
Those chips are not widely adopted. Could be a dead end.

kellogsgit, 2017-06-27:
Here it is 👍 http://www.ebay.com/itm/OBDLink-SX-ELM327-compatible-USB-Diagnostic-Interface-OBDWiz-Software-/161932865237

kellogsgit, 2017-06-27:
OBDLink SX USB is probably the best choice.
Sells 30USD on ebay.com in the US and around 40 EUR in France.
http://www.ebay.fr/itm/OBDLink-SX-USB-diagnostic-professionnel-16-bits-special-Multiecuscan-RENOLINK-/282275106650?hash=item41b8e84f5a:g:EeMAAOSwa~BYPyBW

kellogsgit, 2017-06-27:
If you donate one ODBLinux SX USB, I can implement the required objects in Python OBD. Just answer this post and donate 45 EUR.