brendan-w/python-OBD #44: Autodetect baudrate

Mark-Hetherington, 2016-04-14:
Currently calling:
connection = obd.OBD()
Errors on my device, with some kind of decoding error. Setting the baudrate allows a connection. However I believe it should be possible to autodetect the baudrate, by cycling through the possible baudrates.

I hope to implement this and create a pull request in the near future.

Mark-Hetherington, 2016-04-14:
Close as duplicate of #26.

closed 2016-04-14 by Mark-Hetherington

brendan-w, 2016-04-14:
Yup, got it, thanks though! There's an ELM command for testing baudrate divisors, so I'll be using that.

Mark-Hetherington, 2016-04-14:
If I'm not mistaken a connection at the correct baud rate would have to already be established to issue that command though? I suppose once connected, switching to the fastest available baud rate would be nice.

brendan-w, 2016-04-14:
I added some notes and a reference implementation to the original issue