brendan-w/python-OBD #47: Usage of 'Units'

bendrick92, 2016-05-28:
I'm currently getting responses as follows:

```
obd_response = self.obd_connection.query(obd.commands.INTAKE_PRESSURE)

if not obd_response.is_null():
    return str(obd_response.value) + " " + str(obd_response.unit)
else:
    return "Empty OBD response"
```

The result is being output in kPa, but I'm wanting to convert it to PSI.

Looking at the docs, the [Units](http://python-obd.readthedocs.io/en/latest/Responses/#units) section outlines the string enum, but I'm wondering if there's an existing function to do this conversion?

brendan-w, 2016-05-29:
Python-OBD doesn't currently do any unit conversions. However, I too, have felt a hankering to make this more unit-friendly. Maybe it's time to start researching python unit conversion/annotation libraries. Not sure that I want python-OBD itself to be doing the conversion, but just using a converter's data types would go a long way. I can look into it.

brendan-w, 2016-06-07:
How do people feel about [pint](http://pint.readthedocs.io/en/0.7.2/). So far, it's the most fully-featured unit handling lib I've found, and it has a sane API. Pint values work like this:

```python
>>> v   # response.value
<Quantity(88, 'mph')>

# converts to strings nicely
>>> str(v)
'88 mph'

# handles conversions nicely
>>> v.to('kph')
<Quantity(141.62227199999998, 'kph')>

# you can always pull out the raw python datatype from under the hood
>>> v.magnitude
88

# handles scaler math like normal
>>> v / 2
<Quantity(44.0, 'mph')>

# but, to do non-scaler math, you'll need to specify the unit yourself.
# I see this being the major difference that could put some people off
>>> v + 12
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/brendan/.virtualenvs/obd/lib/python3.4/site-packages/pint/quantity.py", line 587, in __add__
    return self._add_sub(other, operator.add)
  File "/home/brendan/.virtualenvs/obd/lib/python3.4/site-packages/pint/quantity.py", line 503, in _add_sub
    raise DimensionalityError(self._units, 'dimensionless')
pint.errors.DimensionalityError: Cannot convert from 'mph' to 'dimensionless'

# the correct way. Obviously, we'd expose the magical `r` value (pint's unit registry)
# in place of python-OBD's existing `units` enum
>>> v + (12 * r.mph)
<Quantity(100, 'mph')>

# but, the benefit of making people specify those units :)
>>> v + (12 * r.kph)
<Quantity(95.456454306848, 'mph')>

# and, one last tidbit
# the unit registry does string parsing
>>> r("88 mph")
<Quantity(88, 'mph')>
```

I think this is a good way to go, so long as we document the heck out of that `v.magnitude` bit for getting the raw value (no units). While I see most people just printing these values to screens or files, I just don't want to bury the core `int` or `float` beyond the user's reach.

bendrick92, 2016-06-09:
Looks like a great solution!

brendan-w, 2016-06-29:
implemented on devel in 8790148

closed 2016-07-04 by brendan-w