brendan-w/python-OBD #48: Communicate using pybluez rfcomm_client

manikawnth, 2016-06-07:
How do I send commands using a simple rfcomm client socket.
I'm using pybluez and this is my code:

    from bluetooth import *
    sock=BluetoothSocket( RFCOMM )
    sock.connect((host, port))
    sock.send(???)         #what do i need to send here?
    sock.recv(???)         #best way to receive
    sock.close()

brendan-w, 2016-06-11:
Python-OBD [*should*](https://github.com/brendan-w/python-OBD/blob/master/obd/utils.py#L141) natively attach to rfcomm devices. As long as you have bluez configured and paired correctly with the adapter, it should show up as a simple RS232 serial device in `/dev`. To make this lib work with a different backend would require changing [elm327.py](https://github.com/brendan-w/python-OBD/blob/master/obd/elm327.py) to use pybluez instead of pyserial.

alistair23, 2018-11-22:
Closing this issue as Bluetooth connections are supported

closed 2018-11-22 by alistair23