brendan-w/python-OBD #49: Not working with Python 3

thejacko12354, 2016-06-15:
With Python 3.4.2 I only get this
```
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    obdhandler = CObdHandler.CObdHandler()
  File "/home/pi/CarPi/CObdHandler.py", line 12, in __init__
    self._connection = obd.OBD()
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 55, in __init__
    self.__connect(portstr, baudrate, protocol) # initialize by connecting and loading sensors
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 78, in __connect
    if self.port.status >= OBDStatus.ELM_CONNECTED:
TypeError: unorderable types: method() >= str()
```
When using python 2.7.9 it works.
But the Problem is I need python3 for all the other components to work .

Any Ideas?

brendan-w, 2016-06-16:
fixed in 6631c0c. Will release a patched version soon. You can avoid the problem for now by specifying a port string manually, or using `scan_serial()` yourself:

```python
connection = obd.OBD("/dev/ttyUSB0") # create connection with USB 0

# OR

ports = obd.scan_serial()       # return list of valid USB or RF ports
print ports                    # ['/dev/ttyUSB0', '/dev/ttyUSB1']
connection = obd.OBD(ports[0]) # connect to the first port in the list
```

thejacko12354, 2016-06-16:
Thanks a lot!
I'll test it out as soon as I can.

brendan-w, 2016-07-04:
fix released in v0.5.1

closed 2016-07-04 by brendan-w