brendan-w/python-OBD #50: Async TypeError: Odd-length string

chiuwe, 2016-06-15:
When I use async i get the following thread exception when I turn off my car after running a track session (20+ mins)
during my development period, when I drive around the block i don't get this exception when i turn off my car.

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/obd/async.py", line 214, in run
    r = super(Async, self).query(c, force=True)
  File "/usr/local/lib/python2.7/dist-packages/obd/obd.py", line 240, in query
    messages = self.port.send_and_parse(cmd_string)
  File "/usr/local/lib/python2.7/dist-packages/obd/elm327.py", line 312, in send_and_parse
    messages = self.__protocol(lines)
  File "/usr/local/lib/python2.7/dist-packages/obd/protocols/protocol.py", line 184, in __call__
    if self.parse_frame(frame):
  File "/usr/local/lib/python2.7/dist-packages/obd/protocols/protocol_legacy.py", line 50, in parse_frame
    raw_bytes = bytearray(unhexlify(raw))
TypeError: Odd-length string
```

brendan-w, 2016-06-16:
Thanks for the ticket. Heard about this, but never saw a cause. Fix is on master and will be released in 0.5.1 soon.

Jean--Michel, 2016-06-23:
Hi, I also saw this problem, so I apply the patch you commited on protocol_can.py.

Now I have this problem:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/obd-0.5.0-py2.7.egg/obd/async.py", line 214, in run
    r = super(Async, self).query(c, force=True)
  File "/usr/local/lib/python2.7/dist-packages/obd-0.5.0-py2.7.egg/obd/obd.py", line 240, in query
    messages = self.port.send_and_parse(cmd_string)
  File "/usr/local/lib/python2.7/dist-packages/obd-0.5.0-py2.7.egg/obd/elm327.py", line 312, in send_and_parse
    messages = self.__protocol(lines)
  File "/usr/local/lib/python2.7/dist-packages/obd-0.5.0-py2.7.egg/obd/protocols/protocol.py", line 206, in __call__
    if self.parse_message(message):
  File "/usr/local/lib/python2.7/dist-packages/obd-0.5.0-py2.7.egg/obd/protocols/protocol_can.py", line 249, in parse_message
    mode = message.data[0]
IndexError: bytearray index out of range
```
The problem appear on real car, not on the OBDII simulator.
It occurs after the following debug events:
```
[obd] ========================== python-OBD (v0.5.0) ==========================

[obd] Using scan_serial to select port
[obd] Available ports: ['/dev/rfcomm0']
[obd] Attempting to use port: /dev/rfcomm0
[obd] Opening serial port '/dev/rfcomm0'
[obd] Serial port successfully opened on /dev/rfcomm0
[obd] write: 'ATZ\r\n'
[obd] wait: 1 seconds
[obd] read: '\r\rELM327 v2.1\r\r'
[obd] write: 'ATE0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATH1\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATL0\r\n'
[obd] read: 'OK\r\r'
[obd] write: 'ATSP0\r\n'
[obd] read: 'OK\r\r'
[obd] write: '0100\r\n'
[obd] read: 'SEARCHING...\r7E9 06 41 00 80 00 00 11 \r7E8 06 41 00 98 3B 40 11 \r\r'
[obd] write: 'ATDPN\r\n'
[obd] read: '6\r\r'
[obd] Connection successful
[obd] querying for supported PIDs (commands)...
[obd] Sending command: 0100: Supported PIDs [01-20]
[obd] write: '01002\r\n'
[obd] read: '7E9 06 41 00 80 00 00 11 \r7E8 06 41 00 98 3B 40 11 \r\r'
[obd] Sending command: 0120: Supported PIDs [21-40]
[obd] write: '01202\r\n'
[obd] read: '7E9 06 41 20 00 00 00 00 \r7E8 06 41 20 A0 00 00 00 \r\r'
[obd] Sending command: 0220: DTC Supported PIDs [21-40]
[obd] write: '02202\r\n'
[obd] read: 'NO DATA\r\r'
[obd] finished querying with 32 commands supported
[obd] =========================================================================
[obd] '0102: Freeze DTC' is not supported
[obd] Watching command: 0104: Calculated Engine Load
[obd] subscribing callback for command: 0104: Calculated Engine Load
[obd] Watching command: 0105: Engine Coolant Temperature
[obd] subscribing callback for command: 0105: Engine Coolant Temperature
[obd] '0106: Short Term Fuel Trim - Bank 1' is not supported
[obd] '0107: Long Term Fuel Trim - Bank 1' is not supported
[obd] '0108: Short Term Fuel Trim - Bank 2' is not supported
[obd] '0109: Long Term Fuel Trim - Bank 2' is not supported
[obd] '010A: Fuel Pressure' is not supported
[... 82 lines cut ...]
[obd] '015C: Engine oil temperature' is not supported
[obd] '015D: Fuel injection timing' is not supported
[obd] '015E: Engine fuel rate' is not supported
[obd] '015F: Designed emission requirements' is not supported
[obd] Starting async thread
[obd] Sending command: 0112: Secondary Air Status
Started
[obd] write: '01122\r\n'
[obd] read: '7E8 03 41 12 04 \r\r'
1466295130.81	AIR_STATUS : From the outside atmosphere or off
[obd] Sending command: 0110: Air Flow Rate (MAF)
[obd] write: '01102\r\n'
[obd] read: '7E8 04 41 10 00 00 \r00 \r\r'
[obd] Dropping frame for being odd
1466295130.85	MAF : 0.0
[obd] Sending command: 0123: Fuel Rail Pressure (direct inject)
[obd] write: '01232\r\n'
[obd] read: '788 00 \r41 10 00 \r\r'
[obd] Dropping frame for being odd
[obd] Stopping async thread...
[obd] Async thread stopped
```

Regards,
jml

brendan-w, 2016-06-24:
Hi @Jean--Michel, thanks for the debug data. I implemented a fix on master (same cause, different problem). Turns out there was no length checking being done.

Jean--Michel, 2016-06-24:
Hi @brendan-w ,

Thank you.
I saw a similar problem line 115 of the last revision of protocol_can.py, so i add length check:
```
frame.data = raw_bytes[4:]

# <BEGIN PATCH>
if len(frame.data) == 0:
            return False
# <END PATCH>

# read PCI byte (always first byte in the data section)
#             v
# 00 00 07 E8 06 41 00 BE 7F B8 13
frame.type = frame.data[0] & 0xF0
```
Not sure if my patch is correct, but there no more exception.

Regards,
jm.

brendan-w, 2016-06-26:
This happens at fe5740d? The length checks/tests that I added in those commits *should* guarantee at least one data byte (the lower bound is set for 5 bytes, which should be fine for the slice). Unless I messed up the code, and the tests... Yup, when you get around to it, could you send me the debug data so I have something to test with?

Jean--Michel, 2016-06-29:
Hi @brendan-w,

I probably made a mistake with my previous test.
I just use your last "protocol_can.py" and there is no more exception. It works fine except a lot of "None" values I don't understand.

Thank you very much for this software.

Cheers,
jm

closed 2016-07-04 by brendan-w