brendan-w/python-OBD #51: Can't understand why value is 'None'

Jean--Michel, 2016-06-29:
Hi,

Often, some message have 'None' value.
```
[obd] Sending command: 014E: Time since trouble codes cleared
[obd] write: '014E1\r\n'
[obd] read: '7E8 04 41 4E 44 D8 \r\r'
r.time: 1466358933.57
r.name: TIME_SINCE_DTC_CLEARED
r.value: None
```
But sometimes it works:
```
[obd] Sending command: 014E: Time since trouble codes cleared
[obd] write: '014E1\r\n'
[obd] read: '7E9 04 41 4E 4B 9D \r\r'
r.time: 1466358933.7
r.name: TIME_SINCE_DTC_CLEARED
r.value: 19357
```

I tried to reverse the code to understand, but I failed. Could you please help me if you have some time?

Regards,
jml

brendan-w, 2016-06-29:
Somewhat strange. The reason it's reporting None is the difference between the `7E8` and `7E9` headers. The 8 vs. 9 are refering to ECU 1 vs. ECU 2. Python-OBD filters responses based on which ECU the command is designed for. Normally, Python-OBD only ever works with ECU 1 (the engine controller), but in your case, it looks like either ECUs are allowed to respond to this command.

Even more curious, is why it responds correctly to `7E9`, when it should only listen to `7E8`. This could indicate a problem in my ECU labelling logic, but that's a seperate issue. To fix it, mark the command as ECU.ALL, to prevent the `7E8` response from being filtered out:

```python
import obd
from obd import ECU
obd.commands.TIME_SINCE_DTC_CLEARED.ecu = ECU.ALL
```

Jean--Michel, 2016-06-30:
For your information, I also saw 'None' values with the following messages (and may be some more) :
- TIME_SINCE_DTC_CLEARED
- CONTROL_MODULE_VOLTAGE
- RUN_TIME_MIL
- DISTANCE_W_MIL
- ACCELERATOR_POS_D
- THROTTLE_ACTUATOR
- COOLANT_TEMP
- ACCELERATOR_POS_E
- INTAKE_PRESSURE
- MAF
- PIDS_C
- RUN_TIME_MIL
- EGR_ERROR
- RPM
- COMMANDED_EGR
- FUEL_INJECT_TIMING
- AMBIANT_AIR_TEMP

If you need some debug informations, I'd be happy to help.

Regards,
jm

brendan-w, 2016-07-04:
Yeah, if you could send me the debug data for this failure, it might tell me why it's choosing ECU 2. I tested the choosing logic on your debug from #50, and it worked correctly. Were you using a different car for then?

Jean--Michel, 2016-07-05:
Hi,

I think it's the same car, but I'm not sure, sorry about that.
Please find attached a debug log.
IO also tried with another OBD software, it also detects two  ECU (engine(7E8) and transmission(7E9)) and the same protocol as yours (ISO 15765-4 CAN (11-bit ID, 500 kBit/s). 
This sotfware log is following:
```
- OS version: Windows 7
- Succeed open comport
- comport setting detected
- Interface: ELM327 v1.4
- Vehicule voltage: 12.4V

- Protocol setting used: Automatic (auto-detection)
- Waiting for ELM initialization with ECU. Wait 10 secondes.
- Connecting to vehicule
- Protocol detected:ISO 15765-4 CAN (11-bit ID, 500 kBit/s)
- Scanning vehicule functionalities
- ECU detected at  0x7E9 address - Transmission
- ECU 0x7E9 Mode 1, PID : 80000011
- ECU detected at  0x7E8 address - Engine
- ECU 0x7E8 Mode 1, PID : 983B8013
- ECU 0x7E9 Mode 1, PID : 80018001
- ECU 0x7E8 Mode 1, PID : A019A011
- ECU 0x7E9 Mode 1, PID : 400C0000
- ECU 0x7E8 Mode 1, PID : 44DC0009
- ECU 0x7E8 Mode 1, PID : 00008041
- ECU 0x7E8 Mode 1, PID : 21100010
- ECU 0x7E8 compatible: EOBD
- ECU 0x7E9 compatible: EOBD
- ECU 0x7E9 Mode 2, PID : 58180000
- ECU 0x7E8 Mode 2, PID : 58390001
- ECU 0x7E8 Mode 2, PID : 20000000
- ECU 0x7E9 Mode 2 (Freeze frame) supported
- ECU 0x7E8 Mode 2 (Freeze frame) supported
- ECU 0x7E9 Mode 5 (O2 Sensor) not supported
- ECU 0x7E8 Mode 5 (O2 Sensor) not supported
- ECU 0x7E8 Mode 6, PID : 00000001
- ECU 0x7E8 Mode 6, PID : 80008001
- ECU 0x7E8 Mode 6, PID : 00000001
- ECU 0x7E8 Mode 6, PID : 00000001
- ECU 0x7E8 Mode 6, PID : 00000101
- ECU 0x7E8 Mode 6, PID : 00004001
- ECU 0x7E8 Mode 6, PID : 00000001
- ECU 0x7E8 Mode 6, PID : 20000000
- ECU 0x7E9 Mode 6 (OBDMID) not supported
- ECU 0x7E8 Mode 6 (OBDMID) supported
- ECU 0x7E9 Mode 7 (Not confirmed DTC) supported
- ECU 0x7E8 Mode 7 (Not confirmed DTC) supported
- ECU 0x7E9 Mode 8 (Actuator) not supported
- ECU 0x7E8 Mode 8 (Actuator) not supported
- ECU 0x7E9 Mode 9, PID : 10000000
- ECU 0x7E8 Mode 9, PID : 50200000
- ECU 0x7E9 Mode 9 (Vehicle info) supported
- ECU 0x7E8 Mode 9 (Vehicle info) supported
- ECU 0x7E9 Mode 9 (Vehicle info) supported
- ECU 0x7E8 Mode 9 (Vehicle info) supported
- ECU 0x7E9 Mode 10 (Permanent DTC) not supported
- ECU 0x7E8 Mode 10 (Permanent DTC) not supported
- Scan vehicle specific functions (Not compliant with OBD standard)
- Interface connected and initialized, ready to use.
```

I also join two screenshot of this software in cas it might help.
Please, do not hesitate to tell me how I can help.

Thanks a lot for your time and software.
jml.
[04072016-c4jf.txt](https://github.com/brendan-w/python-OBD/files/347408/04072016-c4jf.txt)
[image]
[image]

brendan-w, 2016-07-27:
I've been mulling this one over for a while now, and have only had one thought; Were you using `Async` to perform those reads? I've seen one case where reads/writes trampled over each other and caused flaky behavior, (though I thought I fixed that). If you were, would you mind trying again with normal `obd.query()` and only a single, known-problematic command?

Other than that, I can't find any references to ECU's being allowed to respond in the place of another. An immediate fix for the problem (sorry I didn't mention this sooner), is to check for empty responses yourself:

```python
if not response.is_null():
    # code...
```

JakubJD, 2019-04-28:
Hi,
I would like to ask, have you already fixed the troubles? I am struggling to connect to the car with elemntary command, but I usually get a answer command is not supported or none.

log file:
Ports: 
['/dev/ttyUSB0']
[obd.obd] ======================= python-OBD (v0.7.0) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: 'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'NO DATA\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
Connection status: 
Car Connected
Supported commands: 
GET_DTC
GET_CURRENT_DTC
CLEAR_DTC
PIDS_A
ELM_VERSION
MIDS_A
ELM_VOLTAGE
[obd.obd] Closing connection
[obd.elm327] closing port
[obd.elm327] write: 'ATZ\r\n'

Can you give me an advi

ce please?

Thanks

Jakub

JakubJD, 2019-06-11:
Hey,

I tried already one more time, still with failure.

[image]

[image]

Have you any ideas please.

Thanks

Jakub

alistair23, 2019-06-17:
@JakubJD There is no version 1.5 of the ELM327, so you are using a knock-off chip. The problem is probably that the knock-off chip doesn't correctly match the protocol. Please use a real ELM327 chip instead.

JakubJD, 2019-06-18:
Hey Alistair,

thanks for your answer. I have found under this link: 
https://www.carscanner.info/choosing-obdii-adapter/

the differences and I would like to choose this: 

https://www.ebay.de/itm/Veepeak-OBDCheck-BLE-OBD2-Bluetooth-Scanner-Auto-OBD-II-Diagnostic-Scan-Tool-for/382722436111?hash=item591c08ec0f:g:jMUAAOSwWk5cN5vP

Can you me give an advice, please?

 I will create a new issue in the future. 

Thanks

Jakub

alistair23, 2019-06-18:
I don't know if that is a good one or not, I can't really give you a recommendation. Just make sure that it uses a real ELM327 and you should be ok.

I have used this one: https://www.sparkfun.com/products/9555. It's actually a STN1110 instead of ELM327, but it is compatible and works for me.

Besides that I have a custom PCB with an ELM327 that I use.

JakubJD, 2019-08-18:
Hi Alistair,

I would like to ask you  one more time, whether you can help me. I bought the OBD Adapter: Obdlink LX Bluetooth: https://www.amazon.de/gp/product/B00H9S71LW/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 , after your advice.

But I have still the problems, that I get the answers, that command is not supported. Have you any idea yet?
Thanks a lot

Jakub

[image]