brendan-w/python-OBD #53: Does v0.6.0 not work in python 2.7.11?

cpl77, 2016-08-01:
C:\Python27\python.exe "D:/Documents/Python Projects/Tkinter/version2.py"
Traceback (most recent call last):
  File "D:/Documents/Python Projects/Tkinter/version2.py", line 4, in <module>
    import obd
  File "C:\Python27\lib\site-packages\obd\__init__.py", line 40, in <module>
    from .obd import OBD
  File "C:\Python27\lib\site-packages\obd\obd.py", line 37, in <module>
    from .commands import commands
  File "C:\Python27\lib\site-packages\obd\commands.py", line 34, in <module>
    from .decoders import *
  File "C:\Python27\lib\site-packages\obd\decoders.py", line 255
SyntaxError: Non-ASCII character '\xe2' in file C:\Python27\lib\site-packages\obd\decoders.py on line 255, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

Process finished with exit code 1

It works fine in python 3.4.4 but I have a previous tkinter project and don't want to convert it to python 3

brendan-w, 2016-08-02:
I used a few unicode chars in [a comment](https://github.com/brendan-w/python-OBD/blob/c11ed0d503f6e53d62e1a83a711b7d98f9054ba8/obd/decoders.py#L255-L264) because it made it more readable. Looks like it backfired. I've added utf-8 coding marks to every file so that this doesn't happen again. Will be releasing as `0.6.1` soon.

What's more interesting is that `pytest` *doesn't catch this*, but doing it manually does. Super strange that a test suite, also written in python, didn't find it. I'll look into it.

brendan-w, 2016-08-02:
Should be all good. `pip install --upgrade obd` to get the fix.

cpl77, 2016-08-02:
It very well could be, thank you for your fast responds. I really enjoy python-obd, keep up the great work.

closed 2016-08-02 by cpl77