brendan-w/python-OBD #54: Request PID of commands ELM_VOLTAGE and ELM_VERSION result in ValueError

OrangeTux, 2016-08-19:
The following code raises an `ValueError`:

```
>>> from obd import commands
>>> commands.ELM_VERSION.pid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/auke/projects/Ted/.env/lib/python3.5/site-packages/obd/OBDCommand.py", line 78, in pid
    return int(self.command[2:], 16)
ValueError: invalid literal for int() with base 16: b'I'
```

The PIDs of the commands `ELM_VERSION` and `ELM_VOLTAGE` are respectively `ATI` and `ATRV`. These values can't be casted to integers.

brendan-w, 2016-08-26:
Hi again, thanks for catching this too!

I've added a hex test before parsing, and added tests in dcd6b783.

While I was there, I noticed that `.mode` and `.pid` return zero in the event of an unknown/unparsable mode or PID. This wasn't ideal (seeing that a PID of zero is perfectly valid), so I've altered it to return `None` in these circumstances.

closed 2016-08-26 by brendan-w

OrangeTux, 2016-08-26:
Thanks for solving it.