brendan-w/python-OBD #55: obd_compliance() and fuel_type() in decoder module raise NameError.

OrangeTux, 2016-08-19:
Both methods are expecting a parameter `_hex`, but the logic is applied to a variable `messages`. 

Because `messages` isn't initialized the variable is `None`. Therefore the following line in both methods
raises a `NameError`:

```
d = messages[0].data
```

brendan-w, 2016-08-26:
Good catch, thanks!

Looks like I didn't even have tests for those. Fixed and made tests in 4336e20e.

closed 2016-08-26 by brendan-w

ketyi, 2017-10-30:
Why there is no new release at https://pypi.python.org/pypi/obd then? I'm still facing with this issue with the latest version there.

brendan-w, 2017-10-31:
This project is currently in limbo because 1) I no longer have a car, 2) no longer have time. Thus, I haven't been able to validate everything in master as a whole, and so have held off from blindly pulling triggers.

Am thinking of scheduling some vacation, of which, I could potentially spend a day or two solving the top issues, and cutting a new release (with a friends car). Thorough testing of the current master would be much appreciated.

ketyi, 2017-10-31:
I have downloaded the master code and tried it with a car and can confirm that this fix works.