brendan-w/python-OBD #56: There is connection to car, but I cannot obtain any data

michalpelka, 2016-09-22:
Hi, 
I am having troubles using sample code. I successfully connect to car, but I cannot obtain any data.
I was testing this on on Opel and Kia, same result.
```
>>>r = connection.query(cmd)
[obd.obd] '010D: Vehicle Speed' is not supported
>>>connection.protocol_name()
'ISO 15765-4 (CAN 11/500)'
```

Regards,
Michał

brendan-w, 2016-09-22:
Hello! Please enable debug mode and send me all of the output:

```python
import obd
obd.logger.setLevel(obd.logging.DEBUG) # enables all debug information
```

flakoash, 2017-02-03:
hi, 
im having the same problem, im testing on a Suzuki Grand Vitara(2016)

in debuging mode i got this:

```
>>> c=obd.OBD()
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '?\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v1.5\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\r83 F1 11 7F 01 11 \r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A5\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=38400 PROTOCOL=5
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '01000\r\n'
[obd.elm327] read: b'?\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
```

Regards, Adrian

obduser, 2017-05-14:
I have the same problem. RPM or SPEED is not supported. 
Any Ideas ?

regards

tharunjoy, 2017-07-22:
try this instead 
connection.query(cmd, force=True)

alistair23, 2019-02-14:
This appears to be related to using a knock-off ELM327 chip as "Elm Electronics never made a v1.5" - https://www.elmelectronics.com/wp-content/uploads/2017/01/ELM327DS.pdf

closed 2019-02-14 by alistair23