brendan-w/python-OBD #58: Trying to connect and fails with: Failed to set baudrate

rmd2, 2016-10-26:
Hello,

I'm trying to make a bluetooth connection with ODBII support. I'm trying to use the sample code as below:
```
import obd
connection = obd.OBD() # auto-connects to USB or RF port
cmd = obd.commands.SPEED # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
```
When I try to make a auto connection with this line:

`connection = obd.OBD() # auto-connects to USB or RF port`

this error happens:

```
[obd.elm327] Failed to set baudrate
[obd.obd] Cannot load commands: No connection to car
[obd.obd] Query failed, no connection available
```

I'm using a Mac OSX 10.12.

Thanks !

sommersoft, 2016-11-12:
Do you have full debug logging turned on? If not, add this `obd.logger.setLevel(obd.logging.DEBUG) # enables all debug information` after `import obd`. Then post the full logging result.

Also, the car must be turned to 'ACC ON' at minimum. This is what gave me baudrate failures when I first started using this.

isonet, 2016-12-19:
Hello, I'm having the same problem!

Here the debug output:

```
['/dev/tty.OBDII-SPP']
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-SPP BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b''
[obd.elm327] Response from baud 57600: b''
[obd.elm327] Response from baud 19200: b''
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
```

No idea what's causing this, I think this is mac related. I would be happy to test any ideas you have!

andrewkouri, 2017-10-09:
+1 on Mac OS 10.12.6

mweitzel2005, 2017-11-10:
I've got a Bluetooth ODB adapter which I have successfully paired through mac OS Bluetooth settings.

When I run a test with debug level logging I get similar output as @isonet:
`
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/tty.OBDII-SPP']
[obd.obd] Attempting to use port: /dev/tty.OBDII-SPP
[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-SPP BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b''
[obd.elm327] Response from baud 9600: b''
[obd.elm327] Response from baud 230400: b''
[obd.elm327] Response from baud 115200: b''
[obd.elm327] Response from baud 57600: b''
[obd.elm327] Response from baud 19200: b''
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
`

During the execution I can see how the Bluetooth settings report the OBD adpater as connected.

When I try to manually set the port and baudrate (9600 or 115200) I get the same results:
`
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/tty.OBDII-SPP BAUD=115200 PROTOCOL=auto
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] ATE0 did not return 'OK'
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
`

jlmcgehee21, 2017-11-24:
On OSX 10.12.6, and was seeing the same issue, but I've found that by increasing the timeout in the [ELM237 class](https://github.com/brendan-w/python-OBD/blob/master/obd/elm327.py#L279), I can get a good connection.  I'll work on a PR to make it configurable.

jlmcgehee21, 2017-11-24:
https://github.com/brendan-w/python-OBD/pull/97

mweitzel2005, 2017-12-07:
@jlmcgehee21 out of interest, what value did work for you for the timeout? Am getting the same errors when using 0.1 (default value), 1 or 10 as values.

cormac70, 2018-09-19:
I'm having the same issue, and want to rule something out before I try and fiddle about.

I have a VAG COM 409.1 USB cable, described as compatible with OBD-II. Is this the same as an ELM327 adapter? It has a red light that goes on when plugged into the car, so something's happening in there, but I get the baud and 'no connection to car' messages as above.

Do I need something more sophisticated than this cable? I can't work it out!

masudap, 2018-10-28:
I had the same problem and came across this thread and also other. Here is a specific solution to solve this problem and may not work always, but for me it worked and i hope it might help others.

I solve the problem as described below in a windows 8.1 OS by the following two steps:

1. after pairing the the pc with the elm327, find out the blue tooth port number manually. 
    i.e If using Windows 8/10, navigate: Right-click Start > Control Panel > In the search box, enter "Bluetooth" then select Change Bluetooth settings. a new window will open and in there click the tab "COM ports". there u will see the listed COM ports and note down the one whose direction is outgoing. for example in my case it lists "COM21" and "COM22"  with the names OBDII SPP and OBD respectively.  and "COM21" was the outgoing one and this i need to use.

2. modify the scan_serial() in in the file utils.py so that it will always return the desired port without scanning anything.

i modify the function scan_serial() as listed below, and in essence i commented out the scanning portion of the code and set the return values as result= ['COM21'] which i found in the previous step in case of my computer. 
the location of the utils.py file in my case was: "C:\Users\your_name\AppData\Local\Programs\Python\Python36\Lib\site-packages\obd"

###############################################################

def scan_serial():
    """ Lists serial port names

        :raises EnvironmentError:
            On unsupported or unknown platforms
        :returns:
            A list of the serial ports available on the system
    """
    if sys.platform.startswith('win'):
        ports = ['COM%s' % (i + 1) for i in range(256)]
    elif sys.platform.startswith('linux') or sys.platform.startswith('cygwin'):
        # this excludes your current terminal "/dev/tty"
        ports = glob.glob('/dev/tty[A-Za-z]*')
    elif sys.platform.startswith('darwin'):
        ports = glob.glob('/dev/tty.*')
    else:
        raise EnvironmentError('Unsupported platform')

    #result = []
    result= ['COM21']
    """for port in ports:
        try:
            s = serial.Serial(port)
            s.close()
            result.append(port)
        except (OSError, serial.SerialException):
            pass
	"""
    return result

####################################################################

apecone, 2018-11-19:
Please take a look at #122 and see if this addresses your "failed to auto set baudrate" problem.  Thanks :)

alistair23, 2018-11-22:
I'm going to close this as fixed by #122. Please re-open if required.

closed 2018-11-22 by alistair23

ulherz, 2020-05-17:
Hello,
I am new to this library and I am trying to connect to my car. I got this response, which I supposed was solved by #122 right? Am I understanding something wrong, or is this still a bug? Could not find anything helpfull yet.
Installed via pip and using two different obd interfaces, which are indeed on COM1 (COM3 resp.) with 9600 bps.

connection = obd.OBD() # auto connect

[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['\\.\\COM1']
[obd.obd] Attempting to use port: \.\COM1
[obd.elm327] Initializing ELM327: PORT=\.\COM1 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 9600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 230400: b'\x7f\x7f\r'
[obd.elm327] Response from baud 115200: b'\x7f\x7f\r'
[obd.elm327] Response from baud 57600: b'\x7f\x7f\r'
[obd.elm327] Response from baud 19200: b'\x7f\x7f\r'
[obd.elm327] Failed to choose baud
[obd.elm327] closing port
[obd.elm327] write: b'ATZ\r'
[obd.elm327] Failed to set baudrate
[obd.obd] Closing connection
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================