brendan-w/python-OBD #59: Connection to Car -

nedmc, 2016-11-02:
Hi,

I am trying to use Python-OBD but having a few difficulties getting started. I am using your example code 

```
import obd
connection = obd.OBD() # auto-connects to USB or RF port
cmd = obd.commands.RPM # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print(response.value) # returns unit-bearing values thanks to Pint
```

I get the following response back with the logger enabled. I noticed towards the end of the block that the write tries this **[obd.elm327] write: '01001\r\n'**
Is this correct operation for the code? Or is this where my problem lies?

Any help would be appreciated as I was trying to debug this but with no avail. 

Cheers

```
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: ''
[obd.elm327] Response from baud 9600: '?\r\r>'
[obd.elm327] Choosing baud 9600
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.2a\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\r48 6B 11 41 00 BE 3E B8 11 CA \r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A3\r\r>'
[obd.protocols.protocol] map ECU 17 --> ENGINE
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=9600 PROTOCOL=3
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '01001\r\n'
[obd.elm327] read: b'?\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
[obd.obd] '010C: Engine RPM' is not supported
```

brendan-w, 2016-11-05:
Hi,

Its likely that the ELM version 1.2 doesnt support the extra digit at the end of the request. Try disabl

nedmc, 2016-11-07:
That seems to have done the trick. Read back the coolant temperature last night and all looks good. Thanks very much for reply.

closed 2016-11-07 by nedmc

ulisesbussi, 2017-10-26:
Had the same issue with  "ELM 327 mini" bluetooth solved deactivating the fast mode! Thanks!

WizOnePlus, 2019-10-08:
Had the same issue and that command did the trick as well. I am sure this happens to everyone using the generic version of the ELM 327 chip. Now to an GUI with gauges which seems impossible with any python GUI library. I am working on a JavaFX version but not sure yet how to connect the data from the sensors. If anyone has a better solution let me know. Thanks,