brendan-w/python-OBD #60: Time interval on Async loop.

Rockybilly, 2016-11-03:
An obd software is often used on devices that require low energy consumption, so the cpu cycles are somewhat important. However, it appears that Async retrieves the data on a thread which cycles at full cpu speed. If a developer needed to limit this speed, he/she needs to override this run method on his script, but if this supplied as a parameter at __init__ of Async, then used with time.sleep() (Or whatever method you think best), I think it would prove useful.

brendan-w, 2016-11-06:
Yes, I've been catering to the "make it as fast as humanly possible" crowd, since OBD is notoriously slow. Shouldn't be too hard. Probably make it a keyword arg like `interval` or something. Would happily look at PRs.

I haven't checked it myself, but it shouldn't be consuming *100%* cpu, since it's spending non-trivial time blocking on IO (I think best case is a ~90 msec turnaround time). Has this been giving you grief? If so, I'm curious to know how egregiously python-OBD ate the CPU.

Rockybilly, 2016-11-06:
I haven't tested Async in the car.  Currenty I am using the regular OBD class because the blocking did not causing problems for now. However I did check how many calls do the loop make. It made 5 per second for a single command. It should not be cpu consuming, but I don't know properties of the ELM device fully. If it can keep up with such speed, there is no problem. Btw, seeing you online, I will add some other issues I encountered.

Rockybilly, 2016-11-09:
Tested Async in the car. I made a gui for myself using pygame. With all the drawing functions(with anti-aliasing) and Async, the cpu usage percentage in the Raspberry Pi was 25-60%. I think those cpu spikes caused by the Pygame, not Async. ELM device also seems to keep up with the communication speed. So everything is perfect. However, I think an interval parameter should still be added for those who may want to use it.