brendan-w/python-OBD #61: Trouble code reading

Rockybilly, 2016-11-06:
Currently testing with obdsim, it appears the latest dtc decoder does not parse the code correctly.
obdsim sends P0103 as you may have looked. The code obtains this as P0300. I took a look at the code.
in decoders.py changing line 392 from this,

`d += message.data[2:]`

to this

`d += message.data[1:3]`

currently appears to solve the problem for the first trouble code. The fact that GET_DTC returns a list, means that it supports multiple trouble codes, but now the program returns single one even with multiple dtc's added. The change I made above may cripple the structure of the code for multiple dtc's if it will be added, so I add this as an issue here.

Rockybilly, 2016-11-10:
Found out that this is caused by obdsim which does not provide DTC_COUNT, not even 00 in it's place. Is this behavior consistent among other vehicles, if so, this may need to be reimplemented. checking the number of total bytes read, may solve the problem of detecting if the device provides dtc count.

Rockybilly, 2016-11-11:
After 3 trouble codes, another frame is added to the response. These frames are not parsed within the program. So none of the trouble code gets to be shown. This is caused by `"Dropping frame in multi-frame response not marked as FF or CF"`. This behavior is observed using obdsim.

kellogsgit, 2017-06-27:
Sorry, my patch does not fix this issue.

barracuda-fsh, 2024-05-28:
I'm a developer that made the pyobd remake, using python-obd library. Just recently we noticed that DTC is showing wrong. This is an important one for the library and my program to be of any use. Can we fix it somehow? I don't know right now how to do it, so all the help is appreciated. 

I maded 

+= message.data[2:]

to this

d += message.data[1:3]

As previously stated now it show the DTC correctly, but only one. Can we fix this? Thanks to every contributor.

barracuda-fsh, 2024-05-30:
I fixed it under pyobd. See my code to fix this python-obd library. I can't do a pull request.