brendan-w/python-OBD #64: 'Serial' has no attribute _in__waiting

bar1792, 2016-11-22:
Good Morning,

I am running into an issue when connecting to an ELM327 Bluetooth OBD2 device from my Raspbian Jessie. I am currently using a Connect.sh to manage connecting to the device that contains

sudo rfcomm connect 0 AA:BB:CC:DD:EE:FF 1 &
DISPLAY=:0 sudo python /home/pi/python/HUD.py
exit 0

When the rfcomm command runs it connects to the device and creates /dev/rfcomm0

In which i believed I could use to communicate with my script.

However, this gives me errors and doesn't work. In the docs for the pyobd library I have found that in the descriptions that are constantly used:
/dev/ttyUSB0

Do I specifically need to setup a /dev/ttyUSB0 or is there some driver I may be missing?

I am tried using my same code on my mac, and it ran fine.

Like in the docs I use the following to setup the obd async connection
ports = obd.scan_serial()
connection = obd.Async(ports[0])

bar1792, 2016-11-22:
Alright, I added a SP profile rebooted, made sure the device was paired and trusted, then connected the device and ran my python script. I get the following:
https://www.dropbox.com/s/352vcs7of3k39gs/Photo%20Nov%2022%2C%2010%2018%2051%20AM.jpg?dl=0
https://www.dropbox.com/s/wbdsf3qh04zabcn/Photo%20Nov%2022%2C%2010%2019%2001%20AM.jpg?dl=0

brendan-w, 2016-11-27:
You have an old version of PySerial. Install pyserial >= 3.0. Not quite sure how this dependency avoided getting satisfied, since it's in the [setup.py](https://github.com/brendan-w/python-OBD/blob/master/setup.py#L28). Did you clone the repo manually?

Once that's done, you should be able to use the constructor with no arguments, since [it should auto-detect rfcomm devices](https://github.com/brendan-w/python-OBD/blob/master/obd/utils.py#L170)

xsveda, 2017-03-24:
Had similar problem. It seems that v0.6.1 does not specify required version of `pyserial` so the one already installed in Rasbian was considered as sufficient. It shouldn't be the issue anymore as current HEAD version already states that correctly in `setup.py]`.
For v0.6.1 just update `pyserial` by:
`pip install pyserial --upgrade`

BaReinhard, 2017-03-24:
I should have commented back in this a long time ago, as you can probably tell I was very new to Python then, I since came to the same conclusion as brendan did, xsveda thanks for the update and the reminder.

closed 2017-03-24 by BaReinhard