brendan-w/python-OBD #66: Issue with python OBD running async command Speed

venki3v, 2016-12-07:
When running  python-obd on async write sends the actual command only during the first update.
obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r\n'
[obd.elm327] read: b'7E8 04 41 0C 10 1C \r\r>'

During the next updates it's not sending the actual command as part of the write statement during next updates

obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '\r\n'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.obd] No valid OBD Messages returned

[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '01001\r\n'
[obd.elm327] read: b'7E8 06 41 00 FF FF FF FF \r\r>'
[obd.obd] Sending command: 0120: Supported PIDs [21-40]
[obd.elm327] write: '01201\r\n'
[obd.elm327] read: b'7E8 06 41 20 FF FF FF FF \r\r>'
[obd.obd] Sending command: 0140: Supported PIDs [41-60]
[obd.elm327] write: '01401\r\n'
[obd.elm327] read: b'7E8 06 41 40 FF FF FF FF \r\r>'
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'7E8 07 46 00 84 00 10 00 00 \r\r>'
[obd.obd] finished querying with 199 commands supported
[obd.obd] ===================================================================
[obd.async] Watching command: 010C: Engine RPM
[obd.async] subscribing callback for command: 010C: Engine RPM
[obd.async] Starting async thread
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '010C1\r\n'
[obd.elm327] read: b'7E8 04 41 0C 10 1C \r\r>'
1031.0 revolutions_per_minute
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '\r\n'
[obd.elm327] Failed to read port
[obd.elm327] read: b''
[obd.obd] No valid OBD Messages returned
None
[obd.obd] Sending command: 010C: Engine RPM
[obd.elm327] write: '\r\n'

rafacpitta, 2017-09-06:
Please tell me if you got a solution for this, I'm experiencing the same issue.

closed 2018-11-09 by alistair23

kotEustachy, 2018-12-22:
Hi so is there  safe way to construct fast commands avoiding this issue ?

alistair23, 2018-12-22:
No, there doesn't seem to be.

alistair23, 2018-12-22:
Just something else worth noting is that the log attached here is old as it ends every command with a '\n'. This no longer happened on the master branch. That might fix the problem for you.