brendan-w/python-OBD #67: attribute error: 'Async' Object has no attribute _ Async__thread

jitenmistry, 2017-01-27:
I have tried running python obd with pihud but i keep getting this error. 

When i run the following code without pihud, after printing first repsonse error Failed to connect to port comes and sometimes its the above attribute error

> import obd
> import time
> 
> connection = obd.Async() # create an asynchronous connection
> 
> def new_value(response):
> 	print(response)
> 
> 
> connection.watch(obd.commands.RPM, callback=new_value)
> 
> connection.start()
> time.sleep(5) # do other work in the main thread
> connection.stop()`

rafacpitta, 2017-09-07:
Please try:

`connection = obd.Async(fast=False)`

closed 2018-11-29 by alistair23

Ircama, 2019-03-03:
When `obd.Async` is called with active OBD2 link and ignition off, after some retry (possibly because a slow adapter link is not released in time and consequently the [status becomes NOT_CONNECTED](https://github.com/brendan-w/python-OBD/blob/master/obd/obd.py#L95)), I happen to find this error:

```
Traceback (most recent call last):
  File "...", line ..., in ...
    self.connection = obd.Async(self.cfg['port'], fast=False, timeout=0.1, delay_cmds=0.4)
  File ".../asynchronous.py", line 51, in __init__
    timeout, check_voltage)
  File ".../obd.py", line 64, in __init__
    check_voltage) # initialize by connecting and loading sensors
  File ".../obd.py", line 98, in __connect
    self.close()
  File ".../asynchronous.py", line 127, in close
    self.stop()
  File ".../asynchronous.py", line 85, in stop
    if self.__thread is not None:
AttributeError: 'Async' object has no attribute '_Async__thread'
```

The error occurs within the [`__init__` method](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L48) of asynchronous.py.

I do not understand why [`self.close() in obd.py`](https://github.com/brendan-w/python-OBD/blob/master/obd/obd.py#L97) calls `self.close()` in asynchronous.py (which in turn calls `self.stop()` that fails because [`self.__thread` is not yet defined](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L82) (in fact it is initialized [after `super(Async, self).__init__`](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L54).

Does [`def close(self) in asynchronous.py`](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L118) override [`def close(self) in obd.py`](https://github.com/brendan-w/python-OBD/blob/master/obd/obd.py#L153)?

Ircama, 2019-03-03:
Anyway, moving [`self.__thread      = None`](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L54) before [`super(Async, self).__init__`](https://github.com/brendan-w/python-OBD/blob/master/obd/asynchronous.py#L50) avoids the exception.

alistair23, 2019-03-03:
Do you want to send a PR?

Ircama, 2019-03-04:
ok I will prepare it in the next days