brendan-w/python-OBD #68: questions on async

venki3v, 2017-01-28:
I have a question how async call back supposed to work.

If i use the below code snippet. 

import obd
import time

connection = obd.Async()

# a callback that prints every new value to the console
def new_rpm(r):
    print r.value

connection.watch(obd.commands.RPM, callback=new_rpm)
connection.start()

# the callback will now be fired upon receipt of new values
If i  use a elm 327 freematic simulator to change engine rpm values multiple times , but  the callback is fired successfuly only the first time, for any subsequent change in rpm its not able to read the value from the serial port 

Will the call back be fired whenever the engine rpm changes. I am using elm 327 simulator to send values for engine rpm, fuel level and fuel pressure.

When I set up a async connection and start sending engine rpm values, call back is fired

brendan-w, 2017-01-29:
I'm going to go out on a limb, and guess that your simulator does not fully support the ELM327 spec. Specifically, the feature where newlines re-execute the previous command. Try again, but with this parameter in the Async constructor:

```python
connection = obd.Async(fast=False)
```

kotEustachy, 2018-12-22:
Hi is there safe way without modifying source code to get fast responses async when ELM not support command rewrite ?

alistair23, 2018-12-22:
The support for the "fast" command is in the ELM327 spec ([page 12](https://www.elmelectronics.com/wp-content/uploads/2017/01/ELM327DS.pdf)) since version 1. So it should work for any real ELM327 adaptor.

If it doesn't it probably means that your adaptor is not fully compliant.

The other option is to make sure you are using the latest master branch, as that no longer adds the trailing '\n' charecter which might help fix the problem.

closed 2018-12-22 by alistair23