brendan-w/python-OBD #69: Monitoring Data then issue command

BaReinhard, 2017-01-31:
I am having a hard time trying to figure out if this is possible or maybe its something I am doing wrong.

A brief background, is that I am trying to monitor can bus data so that I can look for specific commands given based on vehicle speed, i.e. trying to enable Video in Motion.

Essentially, I need to monitor data to create a baseline for what is happening over the can bus when the car is at idle in drive with the brake on. Then I need to monitor data above a given speed threshold.

I am unsure if this is possible with the current software. If someone might point me in the right direction it would be greatly appreciated.

thejacko12354, 2017-02-14:
I don't think this is possible with python-OBD, because it just reads the OBD commands.
On some cars you can't even access all can-buses through the OBD-connector.

There are some Can-Bus Monitors out there but often you will need another interface for it, not just an elm327.

brendan-w, 2017-02-19:
Definitely not with python-OBD, though the ELM327 does have a "monitor" mode, where it proxies all CAN messages according to a mask that you can set. It's not pleasant, but could be done. I'd still recommend an actual CAN sniffer. CANtact?

closed 2017-02-19 by brendan-w