brendan-w/python-OBD #70: is_connected() not functioning properly

thejacko12354, 2017-02-14:


brendan-w, 2017-02-19:
```python
>>> import obd
>>> c = obd.OBD()
>>> c.is_connected()
False
```

Can't replicate, please enable debug before making the connection: `obd.logger.setLevel(obd.logging.DEBUG)`

thejacko12354, 2017-02-19:
So this is my code:
``
import obd
obd.logger.setLevel(obd.logging.DEBUG)

connection = obd.OBD('/dev/ttyUSB0', 38400, "6")

print(connection.protocol_id()+" : "+connection.protocol_name())

print(connection.status())
print(connection.is_connected())

cmd = obd.commands.ELM_VOLTAGE

response = connection.query(cmd) 

print(response.value) 
``

This is the result with the version from pypi (Note: it shows the error from #71): 
``
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'01000\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
6 : ISO 15765-4 (CAN 11/500)
Car Connected
True
Traceback (most recent call last):
  File "portTest.py", line 14, in <module>
    response = connection.query(cmd) # send the command, and parse the response
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 249, in query
    if not force and not self.test_cmd(cmd):
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 230, in test_cmd
    if cmd.mode == 6 and self.interface.protocol_id() not in ["6", "7", "8", "9"]:
  File "/usr/local/lib/python3.4/dist-packages/obd/OBDCommand.py", line 71, in mode
    return int(self.command[:2], 16)
ValueError: invalid literal for int() with base 16: b'AT'
``

And now the version from github:
``
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.elm327] write: b'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'\r\rELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] Sending command: b'0600': Supported MIDs [01-20]
[obd.elm327] write: b'0600\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
6 : ISO 15765-4 (CAN 11/500)
Car Connected
True
[obd.obd] Sending command: b'ATRV': Voltage detected by OBD-II adapter
[obd.elm327] write: b'ATRV\r\n'
[obd.elm327] read: b'13.1V\r\r>'
[obd.decoders] Failed to parse ELM voltage
None
``

alistair23, 2018-11-04:
It looks like this is fixed in master and will be fixed with the next PyPI release.

closed 2018-11-04 by alistair23