brendan-w/python-OBD #71: Command ELM_VOLTAGE results in ValueError

thejacko12354, 2017-02-17:
If I query ELM_VOLTAGE i get a ValueError like #54 
Protocol I'm using iss 6 : ISO 15765-4 (CAN 11/500)
I have the latest version installed via pip3 and I'm running Python 3.4.2 on Raspbian Jessie.

`
Traceback (most recent call last):
  File "portTest.py", line 12, in <module>
    response = connection.query(cmd) # send the command, and parse the response
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 249, in query
    if not force and not self.test_cmd(cmd):
  File "/usr/local/lib/python3.4/dist-packages/obd/obd.py", line 230, in test_cmd
    if cmd.mode == 6 and self.interface.protocol_id() not in ["6", "7", "8", "9"]:
  File "/usr/local/lib/python3.4/dist-packages/obd/OBDCommand.py", line 71, in mode
    return int(self.command[:2], 16)
ValueError: invalid literal for int() with base 16: b'AT'
`
Did I forget something?

brendan-w, 2017-02-19:
was fixed in b94dfad, which is in master, but hasn't been released. I haven't been able to do/test anything with this library, because I don't have a car at the moment :P If you clone/install master, that will fix the problem.

closed 2017-02-19 by brendan-w

thejacko12354, 2017-02-19:
Ok I'll do that.