brendan-w/python-OBD #72: Reading BMS Data/BUFFER FULL Filtering/Torque Pro PID Equations

GilbyDettori, 2017-02-19:
Hey,

Firstly, thanks for the great project!
Next up, I'm trying to use python-OBD to read the BMS data from a Hyundai Ioniq Electric Vehicle.

Unfortunately when trying to pull the PIDS, I'm receiving BUFFER FULL errors.
I don't know if this is an issue report or just an RFI, so I've provided the output below when querying.

I plan on contributing to this codebase shortly, once my friend and I have figured out the necessary code to be able to simply copy/paste Torque Pro equations into your app. Meaning anyone can get the PIDs using the information provided by other users for the Torque Pro app. (example ; https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/blob/master/Ioniq%20EV/extendedpids/Hyundai_Ioniq_EV_BMS_data.csv, i.e. 5th col is the equation).
 - this would make it far easier for people to just copy/paste the equations which are already fully functional within the Torque app.

Thanks again for your epic work to-date, and apologies if this query doesn't belong here in github!
Gilby.

I'm using an ELM327 Bluetooth dongle, connected to a Pi3.

[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: 2101:
7EA10166101FFE00000
7EA21092110B905E402
7EA2200000000AA6800
7EB101E6101000003FF
7EB21083801AB8D8232
7EB220000000A000E02
7EC103D6101FFFFFFFF
7EC217926482648A3FF
7EC22CE0E2914111313
7ED102C6101FFFFF800
7ED210091288D6857F7
7ED220228A402470134
7EE037F2112
BUFFER FULL

kellogsgit, 2017-06-25:
I am new to this project and I suggest that you have a look at obdautodoctor.

This works under Windows, Linux and Mac OS.
Using obdautodoctor, please send the command 2101.
And then paste the log here.

This will allow us to study how buffer can be managed.
Buffer management is also described in the EML327 management.

From my understanding, python-obd should send a "clear buffer" (or equivalent) message every 5 seconds to the EML 327 dongle, in order to acknowledge data reception and clear the log.

On my car, when idle, obdautodoctor send OBD code 0101:
2017-06-24 23:45:33	>>		0101
2017-06-24 23:45:33	<<		83F17A7F011280
2017-06-24 23:45:38	>>		0101
2017-06-24 23:45:38	<<		83F17A7F011280
2017-06-24 23:45:43	>>		0101
2017-06-24 23:45:43	<<		83F17A7F011280
2017-06-24 23:45:49	>>		0101
2017-06-24 23:45:49	<<		83F17A7F011280
2017-06-24 23:45:54	>>		0101
2017-06-24 23:45:54	<<		83F17A7F011280

I wonder whether this is a keepalive code or a real query.
If you make your own test, please paste the result below and I will implement a fix in python-OBD.

kellogsgit, 2017-06-25:
By the way, I am also interested in extending python-OBD to support equations, but not before I dig into the code - in a few days/weeks.

kellogsgit, 2017-06-26:
Try to increase speed. What are you connection parameters?
Could also be a faulty connector with low-quality chip.

tobiasnitschmann, 2017-11-25:
Is there a solution to read the State of Charge vie python-obd and an obd bluetooth dongle?

hokus15, 2019-02-06:
@GilbyDettori Did you manage to read the BMS data from Hyundai Ioniq Electric Vehicle?

GilbyDettori, 2019-02-06:
Heya @hokus15. i did, but parked the work ages ago. I'll see if i can still find the code and send it over.

hokus15, 2019-02-06:
Many thanks @GilbyDettori. I'm a bit stuck right now and it would be very useful! I'm eager to see your code.