brendan-w/python-OBD #74: Error: IndexError: list index out of range

scarabdesign, 2017-03-08:
Getting the following error when checking FUEL_STATUS:
`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/obd/obd.py", line 265, in query
    return cmd(messages) # compute a response object
  File "/usr/local/lib/python2.7/dist-packages/obd/OBDCommand.py", line 97, in __call__
    r.value = self.decode(messages)
  File "/usr/local/lib/python2.7/dist-packages/obd/decoders.py", line 302, in fuel_status
    status_1 = FUEL_STATUS[ 7 - bits[0:8].index(True) ]
IndexError: list index out of range

`

alistair23, 2018-11-08:
What are you connecting to? Does this happen when connecting to a real car?

I have seen this issue on an OBD2 emulator, but [it seems](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_03) that the return value is invalid. If it also happens on a real car we can try bit shifting. That should then still give you data back.

alistair23, 2018-11-08:
It could also be an endianness problem

alistair23, 2018-11-08:
Ok, it really does just seem like the data being reported by the car (or emulator) is wrong here. I think the best bet is just to log it and move on.

alistair23, 2018-11-09:
Fixed in master

closed 2018-11-09 by alistair23

scarabdesign, 2018-12-19:
Thanks, buddy! Though I won't be testing it any time soon as I'm off on another tangent.