brendan-w/python-OBD #75: Unable to get data returned.

HuanYangzz, 2017-03-17:
Hi, I'm trying this Library with a mercedes benz (2006) (Protocol
ISO-14230-4 (KWP2000)
ISO-15765-4 (CAN))
with ELM v1.5 bluetooth

Somehow I get this "Dropped frame for being too long" message and failed to get response.
My Source code is copy from: https://github.com/brendan-w/python-OBD/blob/examples/docs/Examples.md 

Just modified: connection = obd.OBD(fast=False)

output as below:

[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: ''
[obd.elm327] Response from baud 9600: '?\r>?\r>'
[obd.elm327] Choosing baud 9600
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v1.5\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\r000 0F 00 7F 5C 0D 02 00 00 00 \r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A6\r>'
[obd.protocols.protocol_can] Dropped frame for being too long
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=9600 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'000 0F 00 7F 5C 0D 02 00 00 00 \r\r>'
[obd.protocols.protocol_can] Dropped frame for being too long
[obd.obd] No valid OBD Messages returned
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'000 0F 00 7F 5C 0D 02 00 00 00 \r\r>'
[obd.protocols.protocol_can] Dropped frame for being too long
[obd.obd] No valid OBD Messages returned
[obd.obd] No valid data for PID listing command: 0600: Supported MIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
[obd.obd] Sending command: 03: Get DTCs
[obd.elm327] write: '03\r\n'
[obd.elm327] read: b'000 0F 00 7F 5C 0D 02 00 00 00 \r\r>'
[obd.protocols.protocol_can] Dropped frame for being too long
[obd.obd] No valid OBD Messages returned
{'messages': [], 'command': None, 'value': None, 'time': 1489715259.505802}
[obd.obd] '010C: Engine RPM' is not supported
{'messages': [], 'command': None, 'value': None, 'time': 1489715259.506446}
[obd.obd] Sending command: 03: Get DTCs
[obd.elm327] write: '03\r\n'
[obd.elm327] read: b'000 0F 00 7F 5C 0D 02 00 00 00 \r\r>'
[obd.protocols.protocol_can] Dropped frame for being too long
[obd.obd] No valid OBD Messages returned
{'messages': [], 'command': None, 'value': None, 'time': 1489715264.734499}
[obd.obd] '010C: Engine RPM' is not supported
{'messages': [], 'command': None, 'value': None, 'time': 1489715264.735027}

Any clues?

HuanYangzz, 2017-03-17:
Another Question: 
I have tested same coding with a Toyota Camry (2008) (ISO-15765-4 (CAN))
I'm able to get RPM, Speed, Throttle Position but not Fuel level input.
it shows Fuel Level is not supported.
Meanwhile, I get response for Fuel Status stated that "Open loop" "temperature too low" <-- some messages like this...
does it related?

LeVadim, 2018-08-03:
You won't be able to get the fuel level on any of the Toyota Vehicles, their sensors don't support that, even on 2018 vehicles

alistair23, 2019-02-14:
This appears to be related to using a knock-off ELM327 chip as "Elm Electronics never made a v1.5" - https://www.elmelectronics.com/wp-content/uploads/2017/01/ELM327DS.pdf

closed 2019-02-14 by alistair23

Ircama, 2019-02-14:
> You won't be able to get the fuel level on any of the Toyota Vehicles, their sensors don't support that, even on 2018 vehicles

Recent python-OBD versions allow [querying Toyota ICE ECU to get fuel level](https://github.com/brendan-w/python-OBD/pull/125#issuecomment-443490629).