brendan-w/python-OBD #76: No valid pid for listing commands

arashk1996, 2017-03-29:
Hi,

I'm trying to connect to a Ford Falcon 2000 model via usb. Connection is fine (only when i use fast=False). However, I cannot read any results.

Here is the debugging output:
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/ttyUSB0']
[obd.obd] Attempting to use port: /dev/ttyUSB0
[obd.elm327] Initializing ELM327: PORT=/dev/ttyUSB0 BAUD=auto PROTOCOL=auto
[obd.elm327] Response from baud 38400: '\x7f\x7f\r?\r\r>'
[obd.elm327] Choosing baud 38400
[obd.elm327] write: 'ATZ\r\n'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
[obd.elm327] write: 'ATE0\r\n'
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: 'ATH1\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATL0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: 'ATSP0\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'SEARCHING...\rUNABLE TO CONNECT\r\r>'
[obd.elm327] write: 'ATDPN\r\n'
[obd.elm327] read: b'A0\r\r>'
[obd.elm327] ELM responded with unknown protocol. Trying them one-by-one
[obd.elm327] write: 'ATTP6\r\n'
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/ttyUSB0 BAUD=38400 PROTOCOL=6
[obd.obd] querying for supported commands
[obd.obd] Sending command: 0100: Supported PIDs [01-20]
[obd.elm327] write: '0100\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.OBDCommand] 0100: Supported PIDs [01-20] did not recieve any acceptable messages
[obd.obd] No valid data for PID listing command: 0100: Supported PIDs [01-20]
[obd.obd] Sending command: 0600: Supported MIDs [01-20]
[obd.elm327] write: '0600\r\n'
[obd.elm327] read: b'CAN ERROR\r\r>'
[obd.obd] finished querying with 7 commands supported

Any help would be greatly appreciated. 

Cheers,
Ash

grimlockrocks, 2017-04-24:
You may want to try different protocols and see. And make sure the engine is started. I had the same problem with my 2008 Volvo, car cannot be connected without running. Hope it helps!

arashk1996, 2017-04-24:
I have tried that and still no luck :/

brendan-w, 2017-05-14:
If none of the protocols work, and the car is on, then I'd be suspect of hardware. Either there's a physical connection problem somewhere, or the adapter has a problem.

arashk1996, 2017-05-14:
All good, ford au series doesn't use standard obd calls, I bought a Ford obd and it works fine :)

kellogsgit, 2017-06-27:
Could be a bad Chinese copy of the ELM327. If you have a list of custom Ford obd calls, please post the complete log of your ODB calls.

arashk1996, 2017-06-27:
No it works with other cars, it's just with my Ford haha, I don't have a
list but I don't think it'd be hard to find off Google

Crasha, 2017-09-02:
@arashk1996 did you figure out how to connect to your Ford? If so, what protocol and baudrate did you use?

closed 2018-11-04 by alistair23