brendan-w/python-OBD #79: Exception: global name 'Unit' is not defined

chaitanyabjoshi, 2017-05-18:
I am trying to use this version of python-OBD to fetch data from the cars. Connection is established successfully but while fetching the data, the code throws an exception.

`[OBD.obd] ======================= python-OBD (v0.6.1) =======================
[OBD.obd] Explicit port defined
[OBD.obd] querying for supported commands
[OBD.obd] Sending command: 0100: Supported PIDs [01-20]
[OBD.obd] Sending command: 0120: Supported PIDs [21-40]
[OBD.obd] Sending command: 0140: Supported PIDs [41-60]
[OBD.obd] Sending command: 0600: Supported MIDs [01-20]
[OBD.obd] Sending command: 0620: Supported MIDs [21-40]
[OBD.obd] Sending command: 0640: Supported MIDs [41-60]
[OBD.obd] Sending command: 0660: Supported MIDs [61-80]
[OBD.obd] Sending command: 0680: Supported MIDs [81-A0]
[OBD.obd] Sending command: 06A0: Supported MIDs [A1-C0]
[OBD.obd] finished querying with 75 commands supported
[OBD.obd] ===================================================================
[OBD.obd] Sending command: 0134: 02 Sensor 1 WR Lambda Current
20170518 150919 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID O2_S1_WR_CURRENT, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0113: O2 Sensors Present
20170518 150919 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID O2_SENSORS, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0110: Air Flow Rate (MAF)
20170518 150919 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID MAF, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0111: Throttle Position
20170518 150919 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID THROTTLE_POS, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0130: Number of warm-ups since codes cleared
20170518 150919 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID WARMUPS_SINCE_DTC_CLEAR, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0131: Distance traveled since codes cleared
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID DISTANCE_SINCE_DTC_CLEAR, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0140: Supported PIDs [41-60]
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID PIDS_C, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0123: Fuel Rail Pressure (direct inject)
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID FUEL_RAIL_PRESSURE_DIRECT, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 014C: Commanded throttle actuator
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID THROTTLE_ACTUATOR, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 011C: OBD Standards Compliance
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID OBD_COMPLIANCE, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0133: Barometric Pressure
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID BAROMETRIC_PRESSURE, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0141: Monitor status this drive cycle
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID STATUS_DRIVE_CYCLE, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0145: Relative throttle position
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID RELATIVE_THROTTLE_POS, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0135: 02 Sensor 2 WR Lambda Current
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID O2_S2_WR_CURRENT, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 010D: Vehicle Speed
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID SPEED, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 010F: Intake Air Temp
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID INTAKE_TEMP, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 010C: Engine RPM
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID RPM, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 010B: Intake Manifold Pressure
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID INTAKE_PRESSURE, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0121: Distance Traveled with MIL on
20170518 150920 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID DISTANCE_W_MIL, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0120: Supported PIDs [21-40]
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID PIDS_B, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0105: Engine Coolant Temperature
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID COOLANT_TEMP, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0104: Calculated Engine Load
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID ENGINE_LOAD, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0100: Supported PIDs [01-20]
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID PIDS_A, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0149: Accelerator pedal position D
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID ACCELERATOR_POS_D, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0146: Ambient air temperature
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID AMBIANT_AIR_TEMP, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 0142: Control module voltage
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID CONTROL_MODULE_VOLTAGE, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 014A: Accelerator pedal position E
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID ACCELERATOR_POS_E, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 011F: Engine Run Time
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID RUN_TIME, Exception: global name 'Unit' is not defined
[OBD.obd] Sending command: 014F: Various Max values
20170518 150921 | WARNING  | Sensors         | run                       | OBD2 Could not fetch data for PID MAX_VALUES, Exception: global name 'Unit' is not defined`

The log lines in between are my program's which are catching the exception. Tested with 3 different cars, Ford Focus, VW Golf GDI and Audi A3. Same result. Do I need to import anything in my script apart from following?
`from OBD import obd, commands`

Thanks and Regards

chaitanyabjoshi, 2017-05-22:
I could fix the issue by adding following statement to OBDResponse.py.
`from .UnitsAndScaling import Unit`
Somehow, that statement was missed and python 2.7 did not like that.
I don't know how to create a pull request. It's too complicated. Hence keeping the issue open.

kellogsgit, 2017-06-25:
I will submit a pull request for that.

alistair23, 2018-11-04:
It sounds like the issue is fixed, I'm going to close this. Feel free to reopen if the issue isn't fixed.

closed 2018-11-04 by alistair23