brendan-w/python-OBD #80: error message:    '010D: Vehicle Speed' is not supported

shahverdy, 2017-05-27:
When i try to read the speed or RPM of vehicle via OBD2 and python27 (version of OBD package of python is 2.0), a message is appeared:
 **'010D: Vehicle Speed' is not supported** 
please help me what should i do?

`import obd

cmd = obd.commands.SPEED
r = connection.query(cmd)
print(r.value)`

Juniorlimaivd, 2017-06-17:
You are being able to connect with your adapter over serial? If yes, send the ATI command and put here the output, and maybe i can help you :)

kellogsgit, 2017-06-25:
From my understanding, OBD is compulsory in the E.U. in each car.
But it is not compulsory for all vendors to implement the complete SPEC of OBD codes.
Do I get it right?

Juniorlimaivd, 2017-06-25:
I think this is not the case. Vehicle speed and RPM are the basic OBD commands. The problem may be in sync the communication with the ELM327, but only with this information i can't assume this. @shahverdy can you run your script in debug mode, and put the output here? and what is your set up to communicate with the OBD?

shahverdy, 2017-06-26:
I use the serial Port Redirector and the connection was established successfully. Also i use the COM4, 35000 TCP port and  9600 bit per second.
This is the screenshot of the execution:
[image]

kellogsgit, 2017-06-26:
Speed is probably too low. A normal connection speed is 38400.
Try connecting using my fork (or wait i submit a patch) and run:

import obd
connection = obd.OBD()
connection.print_discovered()

This will print all information discovered, including port speed and OBD supported commands.

tharunjoy, 2017-07-22:
Try this.. you may get speed

`import obd

cmd = obd.commands.SPEED
r = connection.query(cmd, force=True)
print(r.value)`

alistair23, 2018-11-04:
It sounds like the issue is fixed, I'm going to close this. Feel free to reopen if the issue isn't fixed.

closed 2018-11-04 by alistair23

zeesh1979, 2020-11-12:
iam facing same issue. putting force=True shows me None value
also connection.print_discovered() gives me error that print discovered not found in connection