brendan-w/python-OBD #82: Feature: Explore()

kellogsgit, 2017-06-25:
Dear friends,

Python-OBD looks great, but I am new to car-testing software, so it is better to ask before doing! 

When connecting to a Renault Espace 4dti, Python-OBD return: "9 codes supported" or something like that. Do you think it would make sense to add a discover() funtion to test precisely which ODB codes/sensors are available in a car. This could also help discover custom OBD codes for each brand and then document them. I can handle coding.

What do you think of this feature?

kellogsgit, 2017-06-26:
Done in my fork.

Now I can use the following script:
```
import obd
connection = obd.OBD("/dev/pts/2")

print ("A connection was made to the ECU of your car:")
print ("Protocole: " + connection.get_protocol_name())
print ("Port name: " + connection.get_port_name())
print ("Port rate: " + connection.get_port_baudrate())
print ("The ECU support the following commands:")
a=connection.get_supported_commands()
for i in a:
    print(i)
```

Will see if I can code this into a class or provide the script.

kellogsgit, 2017-06-26:
This is easier:
```
import obd
connection = obd.OBD("/dev/pts/2")
connection.print_discovered()
```

results in:
```
The following settings were used to connect to the ECU:
Protocole: ISO 15765-4 (CAN 11/500)
Port name: /dev/pts/2
Port rate: 9600
The following OBD commands are supported:
0134: 02 Sensor 1 WR Lambda Current
0135: 02 Sensor 2 WR Lambda Current
0136: 02 Sensor 3 WR Lambda Current
0137: 02 Sensor 4 WR Lambda Current
0130: Number of warm-ups since codes cleared
0131: Distance traveled since codes cleared
0132: Evaporative system vapor pressure
0133: Barometric Pressure
0233: DTC Barometric Pressure
0232: DTC Evaporative system vapor pressure
0231: DTC Distance traveled since codes cleared
0230: DTC Number of warm-ups since codes cleared
0237: DTC 02 Sensor 4 WR Lambda Current
0236: DTC 02 Sensor 3 WR Lambda Current
0235: DTC 02 Sensor 2 WR Lambda Current
0234: DTC 02 Sensor 1 WR Lambda Current
0238: DTC 02 Sensor 5 WR Lambda Current
013A: 02 Sensor 7 WR Lambda Current
0141: Monitor status this drive cycle
012F: Fuel Level Input
012E: Commanded Evaporative Purge
012D: EGR Error
012C: Commanded EGR
012B: 02 Sensor 8 WR Lambda Voltage
012A: 02 Sensor 7 WR Lambda Voltage
0146: Ambient air temperature
0149: Accelerator pedal position D
0148: Absolute throttle position C
0127: 02 Sensor 4 WR Lambda Voltage
0126: 02 Sensor 3 WR Lambda Voltage
0125: 02 Sensor 2 WR Lambda Voltage
0124: 02 Sensor 1 WR Lambda Voltage
0123: Fuel Rail Pressure (direct inject)
0122: Fuel Rail Pressure (relative to vacuum)
0121: Distance Traveled with MIL on
0120: Supported PIDs [21-40]
025C: DTC Engine oil temperature
0138: 02 Sensor 5 WR Lambda Current
015B: Hybrid battery pack remaining life
0129: 02 Sensor 6 WR Lambda Voltage
0128: 02 Sensor 5 WR Lambda Voltage
022A: DTC 02 Sensor 7 WR Lambda Voltage
015C: Engine oil temperature
013D: Catalyst Temperature: Bank 2 - Sensor 1
013E: Catalyst Temperature: Bank 1 - Sensor 2
0600: Supported MIDs [01-20]
010C: Engine RPM
013B: 02 Sensor 8 WR Lambda Current
013C: Catalyst Temperature: Bank 1 - Sensor 1
023C: DTC Catalyst Temperature: Bank 1 - Sensor 1
023B: DTC 02 Sensor 8 WR Lambda Current
023A: DTC 02 Sensor 7 WR Lambda Current
010B: Intake Manifold Pressure
023F: DTC Catalyst Temperature: Bank 2 - Sensor 2
023E: DTC Catalyst Temperature: Bank 1 - Sensor 2
023D: DTC Catalyst Temperature: Bank 2 - Sensor 1
[... 117 lines cut ...]
0220: DTC Supported PIDs [21-40]
0221: DTC Distance Traveled with MIL on
0222: DTC Fuel Rail Pressure (relative to vacuum)
0223: DTC Fuel Rail Pressure (direct inject)
0224: DTC 02 Sensor 1 WR Lambda Voltage
0225: DTC 02 Sensor 2 WR Lambda Voltage
0226: DTC 02 Sensor 3 WR Lambda Voltage
0227: DTC 02 Sensor 4 WR Lambda Voltage
025F: DTC Designed emission requirements
0139: 02 Sensor 6 WR Lambda Current
0156: Long term secondary O2 trim - Bank 1
0157: Short term secondary O2 trim - Bank 2
0154: Evap system vapor pressure
0155: Short term secondary O2 trim - Bank 1
0152: Ethanol Fuel Percent
0153: Absolute Evap system Vapor Pressure
0150: Maximum value for mass air flow sensor
0151: Fuel Type
025E: DTC Engine fuel rate
025D: DTC Fuel injection timing
025B: DTC Hybrid battery pack remaining life
025A: DTC Relative accelerator pedal position
0158: Long term secondary O2 trim - Bank 2
0159: Fuel rail pressure (absolute)
```

kellogsgit, 2017-06-28:
I will add voltage control and ELM version control today. Maybe it is better to quit after discovery, as it is only a utility.

E-genin, 2018-06-06:
Hi, 
Are you using an ELM327 Bluetooth or USB adapter ?

alistair23, 2018-11-29:
Closing as this isn't an issue and there is still a pull request open for the feature.

closed 2018-11-29 by alistair23

guruvishnuvardan, 2021-08-13:
Hi,

I am using USB OB2 Connector, my objective is to get the speed of the Car(Honda City 2015 Model India)

Please find the following Code:

**import obd

connection = obd.OBD("/dev/ttyUSB0")
ports = obd.scan_serial()
print(ports)

cmd = obd.commands.SPEED
r = connection.query(cmd)
print(r.value)

OutPut:
['/dev/ttyUSB0']
[obd.obd] 'b'010D': Vehicle Speed' is not supported
None**

I have made an attempt to print using connection.print_discovered(), Please find the output

    **connection.print_discovered()
AttributeError: 'OBD' object has no attribute 'print_discovered'**

Can you please help me resolve this issue

Note: I have attached the USB OBD Connector Picture
[image]

Thanks
Guru