brendan-w/python-OBD #89: Fund raising for hardware/software for development and reverse engineering

kellogsgit, 2017-06-28:
Dear Brendan and all,

I think we need more hardware to boost Python-OBD development and I propose to make a small fundraising. I propose to raise the money on Paypal and then buy and send hardware using eBay/Amazon. Brendan, would you agree with this fundraising. It is targeted for Python-OBD developpers: Brandan, maybe me (hopefuly) and anyone proposing patches.

Here is the following hardware that (IMHO) would be useful:
* OBDLink SX USB with STN11xx firmware: around 35USD in the US and 45EUR in Europe.
* An official ELM327 cable with CAM switch with latest firmware 2.2, around 50EUR from Romania. A good candidate: http://elm-327.eu/elm_usbhsms
* A CAN bus analyser compatible with Linux and wireshark. A good candidate:
http://skpang.co.uk/catalog/pican-canbus-board-for-raspberry-pi-p-1196.html
* An OBD splitter Y cable to allow connection and listening at the same time. Around 10USD.

We probably also need some official software to allow reverse engineering of CAN codes. So IMHO the basic hardware is around 100EUR for each developer involved in Python-OBD.

Further development will also need some proper hardware/software from vendors to reverse engineer. Again around 100EUR for each brand (I think of Renault, might be cheaper with other brands).

Brenda, please tell if you agree with this fund raison. I can raise the money on my paypal account: paypal-perso@poure.com.

Kind regards,
Jean-Michel Pouré

kellogsgit, 2017-07-01:
There is no review of patches and the project seems stalled, so I prefer to do my own project. Closing.

closed 2017-07-01 by kellogsgit

brendan-w, 2017-10-29:
Apologies, just tuning in again. Thank you for the kind offer. Though, as you correctly stated, I have not had time to maintain this (or any) of the projects I normally work on. Working at Tesla has taken a lot of my free time.

Had I the time, I think the next big thing for python-OBD would be a sister "signal repository" for all known manufacturers. There are a ton of CAN signals floating around the internet for a smattering of different vehicles (mostly reverse-engineered), but there appears to be no organization. If we (the community) were able to centralize this information, we could unlock quite a bit of use from our vehicles. python-OBD only ships with what's in the standard, but we all know that's just the tip of the iceberg.

Thankfully, the hardware end of reading from CAN is largely solved. It's decoding the signals that's the important part.