brendan-w/python-OBD #92: Stuck in ELM327 initializing

rafacpitta, 2017-07-09:
Hi everyone,

I've been trying to connect my ELM327 to my PC (Win10) using Bluetooth and I've tried many obd libraries to try to get this connection on. I'm pretty sure my OBD connector works since when I'm using apps with my Android everything goes fine. But when using python-obd I get stuck at the ELM327 initilizaing as follows:`
>>> import obd
>>> obd.logger.setLevel(obd.logging.DEBUG)
>>> con = obd.OBD('\.\COM4')
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=\.\COM4 BAUD=auto PROTOCOL=auto`

I need to Ctrl+Break because it just stands there for all. I've seen on my PC and there are two ports (COM4 and COM5) so I explicitily defined them. But even not doing it so, it stucks the same way.`
>>> import obd
>>> obd.logger.setLevel(obd.logging.DEBUG)
>>> con = obd.OBD()
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['\\.\\COM4', '\\.\\COM5']
[obd.obd] Attempting to use port: \.\COM4
[obd.elm327] Initializing ELM327: PORT=\.\COM4 BAUD=auto PROTOCOL=auto`

rafacpitta, 2017-09-06:
I needed to created a virtual serial port for the bluetooth adapter and put it as argument of the obd.OBD( ) for it to work.

closed 2017-09-06 by rafacpitta

Jdrinkwater1, 2020-11-12:
Im having the same issue, and I believe I have my serial port set up properly. Not sure though if you could explain how you did it that would be great