brendan-w/python-OBD #93: Error on connection

Innercode, 2017-07-21:
Hi,
I'm trying to connect a brand new Peugeot 308SW via an ELM-327 over bluetooth.

The code:
```
import obd
connection = obd.OBD() # auto-connects to USB or RF port
connection = obd.OBD(protocol="6-ISO15765-4(CAN11/500)") # auto-connects to USB or RF port
cmd = obd.commands.SPEED # select an OBD command (sensor)
response = connection.query(cmd) # send the command, and parse the response
print(response.value) # returns unit-bearing values thanks to Pint
print(response.value.to("mph")) # user-friendly unit conversions
```

The error output without protocol specification:
```
[obd.obd] ======================= python-OBD (v0.6.1) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: ['/dev/rfcomm0']
[obd.obd] Attempting to use port: /dev/rfcomm0
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=auto PROTOCOL=auto
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    connection = obd.OBD() # auto-connects to USB or RF port
  File "/usr/local/lib/python2.7/dist-packages/obd/obd.py", line 58, in __init__
    self.__connect(portstr, baudrate, protocol) # initialize by connecting and loading sensors
  File "/usr/local/lib/python2.7/dist-packages/obd/obd.py", line 79, in __connect
    self.interface = ELM327(port, baudrate, protocol)
  File "/usr/local/lib/python2.7/dist-packages/obd/elm327.py", line 137, in __init__
    if not self.set_baudrate(baudrate):
  File "/usr/local/lib/python2.7/dist-packages/obd/elm327.py", line 265, in set_baudrate
    return self.auto_baudrate()
  File "/usr/local/lib/python2.7/dist-packages/obd/elm327.py", line 293, in auto_baudrate
    response = self.__port.read(1024)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

The output error with protocol specification is almost the same. PROTOCOL is than defined.

What goes wrong here?

Any help is much appriciated!

rafacpitta, 2017-09-06:
I've experiencied this error and the solution I found here https://www.raspberrypi.org/forums/viewtopic.php?p=947185#p947185.

You need to add the SP profile editing this:

`sudo nano /etc/systemd/system/dbus-org.bluez.service`

Then add the compatibility flag -C for adding the SP to sdptool. Modify the lines from the file above to:

`ExecStart=/usr/lib/bluetooth/bluetoothd -C`
`ExecStartPost=/usr/bin/sdptool add SP`

Save and reboot, then I created the serial port this way (little bit different from the link I posted):

`sudo rfcomm bind rfcomm1 AA:BB:CC:DD:EE:FF`

Then, on python:

`connection = obd.OBD('/dev/rfcomm1')`

Try also different rfcomm (rfcomm0, rfcomm1, rfcomm2...) and be sure to release all before binding:

`sudo rfcomm release all`

Innercode, 2017-09-27:
Thanks @rafacpitta 

That did the trick

closed 2017-09-27 by Innercode

hamad12125, 2019-07-27:
can i get help same problem i did the trick didn't work

Paulware, 2020-03-13:
I am also having the same trouble after modifying 
etc/systemd/system/dbus-org.bluez.service

with a pi-4, 4gb and latest raspbian Feb-2020
thanks for your help

dmaunder, 2021-07-18:
In my case, those lines were already at the suggested values, and changing the rfcomm command and trying different ports didnt help, any other thoughts?